denzilferreira/aware-client #1: Android wear module missing

irhs, 2015-03-17:
I had to comment out android-wear module references to build the project. Do you have any plans to include the android-wear module in this repo?

Thanks for making it open source.

cheers,
Shrirang

denzilferreira, 2015-03-18:
> .<embedded image>

The Android Wear package is the same as the phone’s package. We are handling detecting which device it is and adapt the package in runtime to turn off/on features that don’t make sense/are not available on the watch. Simply add a symbolic link of aware-core as aware-wear and it should compile just fine. The manifest merging process takes care of getting the right manifest for the phone and wear packages automatically. When you install AWARE on the phone, it will upload itself to the watch for you.

Best,
Denzil;

closed 2015-03-18 by denzilferreira

irhs, 2015-03-18:
Ah. Thanks!