denzilferreira/aware-client #2: New device is not added to aware_device table in the database

sbobek, 2015-03-18:
When new device was added with the QR-code scanning it was not added to the list of devices.
In the web api website the list of devices is empty and it looks like it constantly loading something, but nothing is ever happening. (I have 19 devices but none of them is displayed in the web api panel)
The other data from device sensors are coming to the database.

denzilferreira, 2015-03-18:
> .<embedded image>

The devices by default sync with the server over Wi-Fi, at X minutes intervals (depends on the setting you have on the study). Have the devices synched their data yet? Thats when they get added to the database.

Denzil;

sbobek, 2015-03-18:
Yes, the devices are synchronized, as I can see other data that are coming from other sensors.
Database that I use was created by the previous version of awareframework, so maybe that's the issue?
I had already several devices in aware_device table, and recently I have added a new device that dose not appear in the table (however - as i mentioned - data from other sensors from this device are coming to the database)

denzilferreira, 2015-03-18:
Yes, I’m not sure if we changed the structure of the aware_device table, thus it can’t sync them back in. Can you try with an empty database just to check if there is a difference in the database schema?

Denzil;

sbobek, 2015-03-19:
Yes, the problem was the table schema. The previous version did not have label column. I have added this column manually and now it works. This however makes me wonder how about other tables? Maybe it would be nice to provide backward compatibility and add something to the aware client that will alter the tables which has wrong schema?

Could you at least provide some kind of diff between tables schemas, so I could change other tables?
I can do this on my own, but maybe this kind of comparison will be useful for other users.

denzilferreira, 2015-03-19:
Yes, we could modify the schema on the server side. However, it might happen that you add more columns yourself on the server and then we don’t know what could be wrong if things don’t work as expected. We added the label to the aware_device table to support device labelling on the dashboard. I think the rest of the core tables have stayed the same actually for a long time.

I’ll look into the schema matching between client/server. This is also needed if we update plugin’s content providers, which also get replicated on the server side too. It might be possible to do it on-the-fly when the client sends the database tables and fields to the server side for replication.

Denzil;

sbobek, 2015-03-20:
I saw that there is an update in Weather plugin and it stopped syncing with database. I saw that you have added snow, rain and other columns. It would be really useful if there was at least some kind of notification that the tables have changed, because otherwise everytime an update comes the developer has to manually check if there were some changes to the tables schema and if there were - manually update the database.

denzilferreira, 2015-03-20:
Yes, it's next on my list of things to do. Sorry for that!

sbobek, 2015-03-20:
OK :) Thanks a lot - that will be extremely useful.

denzilferreira, 2015-03-24:
Fixed. From now on, any changed columns are synched between the client and server.

sbobek, 2015-03-24:
For me it seem  not working :/ The database user have all privileges added, but the openweather plugin is not syncing (i did not check in the database, but in the dashboard  shows that there are no records for this device in openweather plugin since last Thursday -- the plugin update day). Everything else is syncing.

denzilferreira, 2015-03-24:
Can you check on the database itself? That might be the dashboard that is not pulling the row count correctly for some reason.

	Denzil;

sbobek, 2015-03-31:
I have checked in the database, bu the table schema for open-weather plugin did not changed. Thus, data is not syncing.

denzilferreira, 2015-03-31:
Thanks for checking. I’ll see what could be missing on the server side to make this happen :-)

Denzil;

denzilferreira, 2015-04-20:
This should be working now. At least it is working on aware’s server. Check if you have the ALTER permission on your database user.

Denzil

    Denzil;—

University of Oulu

Office: TS388 | Team: comag.oulu.fi

@denzilferreira | denzilferreira.com | awareframework.com

sbobek, 2015-04-21:
I have "ALL" permissions added to the user, but it still does not seem to work :/

denzilferreira, 2015-04-22:
OK, so we had some issue with the PHP version running on the server and CodeIgniter (what the web services use for implementation). Can you check now if it works?

Thanks for your patience.

    Denzil;—

University of Oulu

Office: TS388 | Team: comag.oulu.fi

@denzilferreira | denzilferreira.com | awareframework.com

sbobek, 2015-04-22:
I have checked, but it does not seem that it works. I will reboot the aware on my phone and give it some time to synch and let you know tomorrow.

denzilferreira, 2015-04-22:
Just to be sure, you are using the latest openweather plug-in, right?

sbobek, 2015-04-23:
OK. So Here is how it looks like (good you've asked about the latest version of the plugin).
1.) I have checked the SQLite database on the phone, and it is not updated. The plugin is collecting data, but it uses the old schema.
2.) I got an information about the OpenWeather  plugin update from the system notification, so I did not checked that before (Current version that is displayed in application manager is 1.3).
3.) Now, the problem seems obvious: a) either my plugin is not updated  - how do I update it? b) Or my plugin is updated, but somehow works on the old database.

denzilferreira, 2015-04-23:
Try removing and reinstalling the plug-in. That should update the schema locally on the phone. However, you will lose previous collected data :( do a backup of the collected data before doing this. I'll have to implement a similar updating schema function on Android. For now all it does is drop the table and create a new one when updating the schema.
On the server side, the update should keep previous data and modify the database schema to match your phone's schema.

        Denzil;
----
University of Oulu
Group: comag.oulu.fi
Research: denzilferreira.com | awareframework.com

denzilferreira, 2015-04-23:
Now that I’m looking into this… you might actually have to delete the .db file from the AWARE folder for open weather to recreate the database, with the correct schema. This should have been done automatically, but it seems I forgot to change the database version number, hence Android did not update the schema when updating the plugin.

    Denzil;—

University of Oulu

Office: TS388 | Team: comag.oulu.fi

@denzilferreira | denzilferreira.com | awareframework.com

sbobek, 2015-04-23:
OK, so I have updated the plugin, I check the schema of my Local database and it seems OK:
``` sql
sqlite> .schema plugin_openweather
CREATE TABLE plugin_openweather (_id integer primary key autoincrement,
timestamp real default 0,device_id text default '',city text default '',
temperature real default 0,temperature_max real default 0,
temperature_min real default 0,unit text default '',humidity real default 0,
pressure real default 0,wind_speed real default 0,wind_degrees real default 0,
cloudiness real default 0,rain real default 0,snow real default 0,sunrise real default 0,
sunset real default 0,weather_icon_id integer default 0,weather_description text default '',UNIQUE(timestamp,device_id));
```
Good new are also that I did not lose old data - columns in old data are filled with zeros.
But the MySQL database had still old schema :/

I dropped the table from MySQL database on my server and after a few minutes it synced the openweather plugin, but only the new data. How can I force aware to sync the old data too ?

denzilferreira, 2015-04-23:
By default, the synching only happens from when you joined the study date onwards. So probably that’s why you are not getting the old records inserted. When the MySQL database was recreated, does it contain the new columns? I’m wondering… do you have other device sending data that might still have the old version of the plugin? I’m now thinking that this might become a problem (an old version vs a new version) will keep changing the server-side database...

    Denzil;—

University of Oulu

Office: TS388 | Team: comag.oulu.fi

@denzilferreira | denzilferreira.com | awareframework.com

sbobek, 2015-04-23:
OK. So probably that's why i do not have the old data. The MySQL tables were recreated correctly, and new data are now synchronizing OK. The only issue was the old data, but that's fine - I can import it from backup. 
All in all - I am not closing this issue, as there are some problems with that case. The walkaround that worked for me was: in case when the new version of plugin uses new table schema  - remove the MySQL table from the server, and after new table is created, restore historical data from backup.The new data should arrive correctly.

denzilferreira, 2015-04-23:
I’ve just implemented the upgrade function on the client-side, so we don’t lose old data upon database upgrades. It will be included on the next build.

On the server side, I’ll investigate what could be breaking the automatic schema migration.

Thanks for looking into this, this is a complex challenge :-)

    Denzil;—

University of Oulu

Office: TS388 | Team: comag.oulu.fi

@denzilferreira | denzilferreira.com | awareframework.com

denzilferreira, 2015-07-07:
I've fixed this issue yesterday. I had a bug in the databasehandler which would not run an upgrade on the database. Everything should work now, as long as the changes are done primarily on the client side. We don't keep track of changes done on the server side database. As long as the database contains at least the same columns as the server, it should still sync.

closed 2015-12-14 by denzilferreira