denzilferreira/aware-client #4: Aware plugins libraries in maven repository

bambalooon, 2015-03-27:
Could you add Aware plugins libraries to maven repository? It would be really useful for people that want to build plugins based on existing ones.

And also other thing - could you remove from aware-core and plugins git repositories: .idea directory, local.properties, build directories and other that are dependent on programming environment? 
This files aren't necessary for people that want to use source code, but they make updates really time consuming.

denzilferreira, 2015-04-22:
The only issue with the sharing the aware plugins as a library in MavenCentral is that we don't manage some of the contributed plugins source code. It should be the plugin's authors' responsibility to make it available in MavenCentral.

I've removed the extra files from GitHub and should be faster to download the sources.

bambalooon, 2015-04-22:
Thank you for your response!

Actually I'm interested (as I understand) only in Your plugins: Google Activity Recognition, Google Fused Location, OpenWeather.

Thank you for removing this files, it will speed up my development.

denzilferreira, 2017-07-10:
Solved in the latest. They can be retrieved directly from github using jitpack.io :-)

closed 2017-07-10 by denzilferreira