denzilferreira/aware-client #6: [BUG] Inserting some polish characters to database fails (probably not only polish)

bambalooon, 2015-05-05:
As I pointed out in title - when you try to synchronize local Aware DB with some polish characters ('ę' for sure) - nothing happens, you don't get any failure - response status code is 200(OK).
Other polish characters (like 'ł' for example) are replaced with question mark ('?').
I think Aware should correctly replicate characters from every language.

It would be also nice to change database default charset to utf8 from latin1_swedish_ci which can generate some additional issues.

denzilferreira, 2015-05-06:
This should be solved now. The default charset has been set to utf8 system-wide. However, if you are hosting your database, it should use your database’s default charset.

Cheers,
Denzil

bambalooon, 2015-05-06:
Thanks for quick response.
After Your fix I no longer get failure on synchronization, but unfortunately polish characters are still replaced (and not only polish).

For example characters: 'ę', 'ł', 'ż', 'ś' are replaced with '?' and characters 'ó', '&#189;' are removed from text.

denzilferreira, 2015-05-06:
> characters 'ó', '½' are removed from text.

OK, made some more configuration changes on how we handle the encoding. 
Try again and let us know if it's fixed. Thanks for helping us improve 
AWARE :) Really appreciate it!

bambalooon, 2015-05-07:
It's still the same. Polish characters doesn't work.
And I'm glad You are this active in fixing bug that is stopping my progress.

denzilferreira, 2015-05-09:
OK, try now. Hopefully this should be solved now.

bambalooon, 2015-05-13:
Unfortunately - no changes.

denzilferreira, 2019-10-01:
This is solved in mysql configuration, where the default encoding is set.

closed 2019-10-01 by denzilferreira