denzilferreira/aware-client #7: [BUG] DB synchronization in batch

bambalooon, 2015-05-06:
When I try to synchronize DB with Aware server for some reason it fails (there is not feedback about exception and no change in remote DB).

It's a json that is posted as 'data' in insert request:
```json
[{
	"timestamp": 14306816639010,
	"device_id": "b0c46dda-37a8-4a50-9740-0655e17d5fe3",
	"double_poi_id": 534590614,
	"type": "node",
	"double_lat": 50.0142059,
	"double_lon": 19.9279012,
	"name": "Galeria Solvay",
	"shop": "supermarket"
},
{
	"timestamp": 14307555075910,
	"device_id": "b0c46dda-37a8-4a50-9740-0655e17d5fe3",
	"double_poi_id": 2708916232,
	"type": "node",
	"double_lat": 50.0435547,
	"double_lon": 19.9439609,
	"name": "Klopsztanga – trzepak reaktywacja"
},
{
	"timestamp": 14307555214700,
	"device_id": "b0c46dda-37a8-4a50-9740-0655e17d5fe3",
	"double_poi_id": 1500744003,
	"type": "node",
	"double_lat": 50.0290285,
	"double_lon": 19.9613803,
	"name": "Pomnik Ofiar Faszyzmu",
	"trsm": "viewpoint"
},
{
	"timestamp": 14307560115450,
	"device_id": "b0c46dda-37a8-4a50-9740-0655e17d5fe3",
	"double_poi_id": 1480534372,
	"type": "node",
	"double_lat": 50.0429659,
	"double_lon": 19.9468393,
	"name": "Basen KS Korona",
	"amnt": "swimming_pool"
}]
```

When I send every row separately everything works!

denzilferreira, 2015-05-06:
If you are doing a batch insert, the elements need to have exactly the same
columns. It seems you are inserting data to different tables in one go,
since I see different column names on the array.

bambalooon, 2015-05-06:
Update: This issue is no longer present after I created new study with utf8 db.
And about my insert - it's insert to single table some columns are omitted for some rows because they're null.

closed 2015-05-06 by bambalooon