denzilferreira/aware-client #8: [BUG] Synchronization of deleted items

bambalooon, 2015-07-05:
Hi,

As I noticed, there is no synchronization for deleted or updated items in Aware.
I understand that synchronizing updates can be difficult and overkill, but sync of deleted items, should be quite easy and manageable. Then users could develop apps using delete-insert operations instead of update.

If it isn't possible, could you give privilege to manage triggers? It'd allow us to create some hacks. I prefer to create hack than mess with Aware code. Changing sync queries to "insert on duplicate update" also would work for my case.

Regards,
K.B.

denzilferreira, 2015-07-05:
Hi there K.B.,

As you mentioned, we only send new data to the server and assume the
historic data (older records) remains the same. This was done to preserve
bandwidth and only sync what is new. If your app is changing previous
records, we don't get these changes.

So the current method used is:
- check on the server what was the last entry on the database
- select on the phone all records after that last entry (timestamp)
- send the data to server

What you could do is update the timestamp on the phone so that it syncs
back to the server. And then we can change the server sync to insert on
duplicate update. This should work, right?

denzilferreira, 2015-07-05:
Now that i think of it, don't you have full permissions on your database?
So in theory you should be able to create triggers to handle this for your
needs.

If you don't have permissions, let me know and I'll adjust them accordingly.

bambalooon, 2015-07-05:
Between this 2 solution I'd prefer trigger. Unfortunately I don't have necessary permissions to use it. I'd be really thankful if you could give me full permissions to my database and also if I'd have full permissions to newly created databases in future.

Regards,
K. B.

denzilferreira, 2015-07-05:
Yes, everyone should have full access to their databases. I'll let you know
when ready!

bambalooon, 2015-07-05:
Thank you very much.

akuvisur, 2015-07-07:
Hey, I added the TRIGGER privileges to all of your existing studies. The privileges will also be included automatically in all future for new studies once the dashboard is updated (most likely by friday this week).

denzilferreira, 2015-07-07:
I've just updated the dashboard to allow this on future studies as well.
Enjoy!

bambalooon, 2015-07-09:
Great thanks!

closed 2015-07-09 by bambalooon