denzilferreira/aware-client #9: Application usage problems

lifedj, 2015-07-15:
I'm trying to use your framework to collect notifications but I'm having some troubles.

Here is the procedure I followed:

1. I downloaded awareframework.apk from this url

http://www.awareframework.com/awareframework.apk

2. I installed it on my smartphone (Karbonn sparkle V) and I opened it to set all the settings
3. I enabled every available sensor (after enabling accessibility service)
4. I exited the app and after a while I came back in the app
5. If I went to the "Applications" tab I found the "Application usage" flag disabled.
If I try to enable it again, the system asks me to enable the accessibility service again, but it is still enabled.

Do you know anything about this problem?
I tried on a Nexus 5, too, and I had the same problem!

denzilferreira, 2015-07-15:
>    1. I installed it on my smartphone (Karbonn sparkle V) and I opened it
>    2. I enabled every available sensor (after enabling accessibility
>    3. I exited the app and after a while I came back in the app
>    4. If I went to the "Applications" tab I found the "Application usage"

Hi there,

I'm unable to replicate this issue. What version of Android do you have?
I'm unsure about incompatibilities with the first phone manufacturer but it
should work just fine on N5.

Let's start from the beginning:
- uninstall AWARE
- delete the AWARE folder that is on your phone's external storage
- download the latest version of AWARE from the website
- install it
- open it once, don't activate anything and close it
- go to Settings > Accessibility Services and turn AWARE on
- open the client and activate the debug messages first
- if you have the phone connected to your computer and Android Studio open,
you should see messages from AWARE on the logcat
- activate just the applications logging
- if it goes well, you will see background and foreground application usage
messages
- do not activate all the sensors at the same time, that is asking for
trouble in terms of battery life

lifedj, 2015-07-15:
Thank you for the reply!
On the nexus 5 there is CyanogenMOD 12.1, instead on Karbonn Sparkle V there is Android 5.1 lollipop (native but with root).

Here you can find the error I obtain in the log:

07-15 12:40:34.546  15542-15542/? D/AndroidRuntime﹕ Shutting down VM
07-15 12:40:34.548  15542-15542/? E/AndroidRuntime﹕ FATAL EXCEPTION: main
    Process: com.aware, PID: 15542
    java.lang.IndexOutOfBoundsException: Invalid index 0, size is 0
            at java.util.ArrayList.throwIndexOutOfBoundsException(ArrayList.java:255)
            at java.util.ArrayList.get(ArrayList.java:308)
            at com.aware.Applications.onAccessibilityEvent(Applications.java:185)
            at android.accessibilityservice.AccessibilityService$1.onAccessibilityEvent(AccessibilityService.java:659)
            at android.accessibilityservice.AccessibilityService$IAccessibilityServiceClientWrapper.executeMessage(AccessibilityService.java:752)
            at com.android.internal.os.HandlerCaller$MyHandler.handleMessage(HandlerCaller.java:37)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5254)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
07-15 12:40:34.550      599-827/? W/ActivityManager﹕ Force finishing activity 1 com.aware/.Aware_Preferences

denzilferreira, 2015-07-15:
OK, we'll take a look.

NBerkel, 2015-07-15:
Hello lifedj,

I have been looking into the problem you describe and can replicate your problems. From my analysis I believe that the accessibility status of AWARE is reported incorrectly - resulting in AWARE continuously asking you to enable AWARE in the accessibility settings.

Could you install the following .apk and check if the problem persists for you? This is only a temporary fix, but hopefully it can help you on your way until we can figure out a more permanent solution.

https://dl.dropboxusercontent.com/u/1964009/aware-core-phone-release.apk

Let us know if it helps. :)

(Please remember to uninstall your current version of AWARE first, and to enable AWARE in accessibility settings again after installing the new .apk.)

lifedj, 2015-07-16:
Thank you,
I've tried it only on the Karbonn Sparkle V and, actually, it does not crash.

But, looking inside the db generated files, notifications are not yet stored.

Here you can find what I find in logcat whenever a Whatsapp notification arrive:

07-16 12:57:14.596   1109-11845/? I/GCM﹕ GCM message com.whatsapp 0:1437044218309826%a3bffb5f00000031
07-16 12:57:15.350      168-168/? I/main﹕ type=1400 audit(0.0:85): avc: denied { getattr } for path="socket:[2822]" dev="sockfs" ino=2822 scontext=u:r:zygote:s0 tcontext=u:r:init:s0 tclass=unix_stream_socket permissive=1
07-16 12:57:15.414      601-617/? I/ActivityManager﹕ Start proc 12545:android.process.acore/u0a3 for content provider com.android.providers.contacts/.ContactsProvider2
07-16 12:57:15.629  12545-12563/? I/ContactLocale﹕ AddressBook Labels [it-IT]: [, A, B, C, D, E, F, G, H, I, J, K, L, M, N, O, P, Q, R, S, T, U, V, W, X, Y, Z, Α, Β, Γ, Δ, Ε, Ζ, Η, Θ, Ι, Κ, Λ, Μ, Ν, Ξ, Ο, Π, Ρ, Σ, Τ, Υ, Φ, Χ, Ψ, Ω, , А, Б, В, Г, Д, Ђ, Е, Є, Ж, З, И, І, Ї, Й, Ј, К, Л, Љ, М, Н, Њ, О, П, Р, С, Т, Ћ, У, Ф, Х, Ц, Ч, Џ, Ш, Щ, Ю, Я, , א, ב, ג, ד, ה, ו, ז, ח, ט, י, כ, ל, מ, נ, ס, ע, פ, צ, ק, ר, ש, ת, , ا, ب, ت, ث, ج, ح, خ, د, ذ, ر, ز, س, ش, ص, ض, ط, ظ, ع, غ, ف, ق, ك, ل, م, ن, ه, و, ي, , ก, ข, ฃ, ค, ฅ, ฆ, ง, จ, ฉ, ช, ซ, ฌ, ญ, ฎ, ฏ, ฐ, ฑ, ฒ, ณ, ด, ต, ถ, ท, ธ, น, บ, ป, ผ, ฝ, พ, ฟ, ภ, ม, ย, ร, ฤ, ล, ฦ, ว, ศ, ษ, ส, ห, ฬ, อ, ฮ, , ㄱ, ㄴ, ㄷ, ㄹ, ㅁ, ㅂ, ㅅ, ㅇ, ㅈ, ㅊ, ㅋ, ㅌ, ㅍ, ㅎ, , あ, か, さ, た, な, は, ま, や, ら, わ, #, ]
07-16 12:57:15.761     601-1389/? I/ActivityManager﹕ Killing 11627:com.fsck.k9/u0a97 (adj 15): empty #17
07-16 12:57:15.954      601-601/? I/ValidateNoPeople﹕ Validating: 0|com.whatsapp|1|null|10087
07-16 12:57:15.954      601-660/? I/ValidateNoPeople﹕ Executing: validation for: 0|com.whatsapp|1|null|10087
07-16 12:57:15.967      601-601/? D/lights﹕ set_led_state colorRGB=FFFFFFFF, onMS=1000, offMS=4000
07-16 12:57:15.968      601-601/? D/lights﹕ blink_green, level=255, onMS=1000, offMS=4000
07-16 12:57:15.972      601-601/? D/lights﹕ write_int open fd=194
07-16 12:57:15.972      601-601/? D/lights﹕ write_int open fd=194
07-16 12:57:16.024      601-660/? I/ValidateNoPeople﹕ final affinity: 0.5
07-16 12:57:18.352  10983-12566/? D/AWARE﹕ Running 35 applications
07-16 12:57:18.592  10983-12566/? I/art﹕ Explicit concurrent mark sweep GC freed 23398(1393KB) AllocSpace objects, 0(0B) LOS objects, 25% free, 7MB/9MB, paused 821us total 118.964ms
07-16 12:57:19.310  10983-12566/? W/ResourcesManager﹕ Asset path '/system/framework/com.google.android.media.effects.jar' does not exist or contains no resources.
07-16 12:57:46.723    2688-2797/? I/3gw.NetworkMonitor﹕ updateTraffic completed in 472 ms

denzilferreira, 2015-07-16:
Did you activate the notification sensor? You should see some feedback when
it is turned on/off.

lifedj, 2015-07-16:
Yes, this is the screenshot:

[image]

denzilferreira, 2015-07-16:
Do you get the notifications on the Nexus 5?
Thanks for helping us debug this.

lifedj, 2015-07-17:
You are right ... my last answer was ambiguous ... Yes: I activated the notification sensor as shown in the screenshot, but the app continues not to store notifications, returning the log shown in the previous message! ;-)

lifedj, 2015-07-21:
I tried again to uninstall and reinstall the aware app, but the problem is still existing: it doesn't save any notification!

I uploaded all the generated log files at the following link (log is included in a txt file):
[awareLog21July.zip](http://lifedj.altervista.org/public/awareLog21July.zip)

Thank you in advance for your effort! ;-)

lifedj, 2015-07-22:
I tried to compile the project on Android Studio by myself and now your client works fine: it collect notifications!!!
You should probably compile the project again and publish the updated apk! ;-)

denzilferreira, 2015-07-22:
Thanks! I'll update it later today.

lifedj, 2015-07-22:
Uhm ... too entusiasm ... It store notifications but crashes as soon as I open the app! What did you do when you sent me the modified apk?

denzilferreira, 2015-07-22:
Just fixed these issues. Try the latest build. I’ve pushed the changes to github too.

lifedj, 2015-07-23:
Perfect, I've just downloaded it and it seems to work without crashes ... I will let you know if it works fine!!!

Thank you very much for your support!!! ;-)

closed 2015-12-14 by denzilferreira