denzilferreira/aware-client #22: Google vision services not operational right now

rkdarst, 2016-06-09:
https://developers.google.com/vision/announcement

This says that there is a bug in the API and it is disabled for new users (and only new users).  I'm filing a issue here to keep track.  The date on the announcement is 27 May 2016 so hope the "several weeks" they say is correct.

denzilferreira, 2016-06-09:
Latest version on Play Store has a fallback for this :) try it!

rkdarst, 2016-06-10:
Hi,

We tested today.  On those phones that already had vision services, it works.  For others, clicking QR button crashes application immediately ("unfortunately, AWARE has stopped").  I don't currently have any other debugging info, unfortunately...

rkdarst, 2016-06-15:
From the log.  Does this help any.  This log found by starting AWARE, clicking QR code button, it immediately crashes, then `adb logcat -d`.  Let me know if more log context is useful or I can get more info somehow.

```
06-15 11:39:26.538 14931 14931 I Vision  : Requesting barcode detector download.
06-15 11:39:26.545  1428 16663 I Vision  : Attempting to open download_details.json in com.google.android.gms.vision
06-15 11:39:26.554  1428 16663 W Vision  : Failed to open download_details.json in com.google.android.gms.vision: android.content.pm.PackageManager$NameNotFoundException: com.google.android.gms.vision
06-15 11:39:26.554  1428 16663 I Vision  : Attempting to open download_details.json in com.google.android.gms.policy_ccocr_vision
06-15 11:39:26.554  1428 16663 W Vision  : Failed to open download_details.json in com.google.android.gms.policy_ccocr_vision: android.content.pm.PackageManager$NameNotFoundException: com.google.android.gms.policy_ccocr_vision
06-15 11:39:26.554  1428 16663 W Vision  : Reading download details from hard-coded Java
06-15 11:39:26.571 14931 14931 D AndroidRuntime: Shutting down VM
06-15 11:39:26.583 14931 14931 E AndroidRuntime: FATAL EXCEPTION: main
06-15 11:39:26.583 14931 14931 E AndroidRuntime: Process: com.aware.phone, PID: 14931
06-15 11:39:26.583 14931 14931 E AndroidRuntime: java.lang.RuntimeException: Unable to start activity ComponentInfo{com.aware.phone/com.aware.phone.ui.Aware_QRCode}: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.support.v4.widget.DrawerLayout.isDrawerOpen(int)' on a null object reference
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2416)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2476)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at android.app.ActivityThread.-wrap11(ActivityThread.java)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:102)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:148)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:5417)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
06-15 11:39:26.583 14931 14931 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.support.v4.widget.DrawerLayout.isDrawerOpen(int)' on a null object reference
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at android.support.v7.app.ActionBarDrawerToggle.syncState(ActionBarDrawerToggle.java:239)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at com.aware.phone.ui.Aware_Activity.onPostCreate(Aware_Activity.java:107)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at android.app.Instrumentation.callActivityOnPostCreate(Instrumentation.java:1188)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2398)
06-15 11:39:26.583 14931 14931 E AndroidRuntime:        ... 9 more
06-15 11:39:26.584   772  3481 W ActivityManager:   Force finishing activity com.aware.phone/.ui.Aware_QRCode
06-15 11:39:26.587   772  3481 W ActivityManager:   Force finishing activity com.aware.phone/.Aware_Client

```

denzilferreira, 2016-06-15:
Yes, working on it. Thanks for sharing!

All my best,
Denzil;

denzilferreira, 2016-06-16:
OK, this should be working on the latest push.

rkdarst, 2016-06-18:
Hi,

It works on my phone and also another tester's phone, so I'm closing this as fixed.  Thanks!

closed 2016-06-18 by rkdarst