denzilferreira/aware-client #23: Android app requires camera, which restricts google play availability

rkdarst, 2016-06-18:
From forum post:

>> – It doesn’t run on my tablet (requires back camera feature?)

> Can you check the logcat and see what could be the reason? I’ve tested it before on a tablet and had no issues.

I should clarify: it isn't that it doesn't run, but that it is not seen as available on google play (the latest version).  I could possibly install it manually, but that restricts our audience.  Steps to reproduce: search "Aware Framework" in google play app on device, or go to google play web and search aware framework, it is listed as "not compatible with $tablet name.

After investigating, I think the reason is the declared required features, since this tablet does not have a back camera.  In aware-phone AndroidManifest.xml:

```
    <uses-feature android:name="android.hardware.camera" />
    <uses-feature android:name="android.hardware.camera.autofocus" />
```

According to https://developer.android.com/guide/topics/manifest/uses-feature-element.html#hw-features, `android.hardware.camera` means back facing camera.

Being able to run on as wide a variety of platforms as possible is important to us, since our subjects may be from demographics which are not as likely to have modern devices.  And our use case would be monitoring all devices, not just the primary device.

Suggested workaround: adding `android:required="false"` to the camera-related permission and if they are not available, the app does not use them.  Instead, one has to configure by manually entering the config URL.

denzilferreira, 2016-06-23:
Fixed in the latest 👍

closed 2016-06-23 by denzilferreira

rkdarst, 2016-09-10:
By the way, this was not fixed at the time, but apparently cf23dda has fixed it.