denzilferreira/aware-client #24: Client pings oulu server regardless of configured webservice

rkdarst, 2016-06-18:
From the forum:

>> Does the android client always ping the Oulu server?

>Yes. We like to keep track of how many active clients there are. For statistics and for us to measure the demand on our server.

In the case I am planning for, we are not using the Oulu server.  Pinging it reveals extra information, and for some of our studies even the fact that someone might be involved would be considered information which shouldn't be shared.  Our ethical permissions state that our apps do not give any information to any third party (and also I wouldn't be comfortable with sharing this information about my subjects).

Perhaps could the client ping the configured webservice instead?  For the google play version, it should have statistics on total installations.

Thanks,

- richard

denzilferreira, 2016-06-23:
The only information with the ping is:
- device id
- current time

We need to know how many devices are using AWARE out there, regardless of what webservice is being used for a study.

rkdarst, 2016-06-23:
I understand the need to know the number of clients for scientific reporting purposes.  I guess at least you have Google Play stats?  For example, leaking initial IP address + persistent device ID makes enough to track a person and infer what study they are in, and through that make inferences about their health.  We also can't get ethics permission to use AWARE with this.

So as it is, we'll need to start working on our own branch.  But we should try to avoid a hard fork as much as possible so that we can still work together.  Are you interested in that?  Any personal preferences for how we should do this to stay in collaboration as much as possible?

Though, let's see about longer term strategic collaborations still (I'll contact you later).

Thanks,

- Richard

rkdarst, 2016-10-03:
Code exists, this will now be optional and subjects can always turn it on if they choose.  Awaiting verification.

denzilferreira, 2016-10-03:
Just merged.

closed 2016-10-03 by denzilferreira