denzilferreira/aware-client #25: [wishlist] configuration mode to allow complete removal of all data

rkdarst, 2016-06-18:
>> Can the sdcard AWARE folder be deleted when app uninstalled?

> No, it can not. It’s intentional. Some participants decide to uninstall the app themselves. If we cleared that folder, the researcher would lose all the data.

In our use case, we could be deploying to hundreds of clients (aside: it will be installed and configured before subjects meet researchers physically).  The study will be long (an academic year at least), and we'd expect some clients to drop out early, and we won't plan on meeting people after the study.  And for our permissions to do the project, and also to minimize the support load, we would like to say "A subject can opt-out of the study at any time by uninstalling the app, when they do so their phone is restored to the state it was before the project started".  In this case, data is collected via the webservice, and if data is not sent, then we don't get it.  (aside: in this use case, the "delete data immediately after sending" option is very useful, since we also want to say that data is stored on the phone for the minimum time possible).

So, from our perspective, we could contribute a hidden option that would put the AWARE folder inside the Android/data folder, which appears to be removed on de-installation.  This is also good because the data is hidden to make things as un-intrusive as possible.  This should be a hidden option that is only available if you link to a server and the server sends it.

What do you think?

Thanks,

- Richard

rkdarst, 2016-09-22:
From email conversation by me/@denzilferreira.  I was investigating how to do this.  Any comments?

>> - we have the data by default stored in external storage
>>  (sdcard/Android/app_name) where it will be removed when app is
>>  uninstalled or user does "clear data".  This was important because
>>  we wouldn't see users at the end of the study.  I was wondering
>>  about making it configurable, but then you get into corner cases
>>  like needing to move data, which I didn't want to deal with yet.
>>  However, we can do this now.  Any ideas?  (Our ethics say
>>  "uninstalling the app returns the phone to the original state”)
>
>We could support this as an option of the client itself, e.g., Upon uninstall, keep or delete data? If the user has set
>to keep the data, we just save a backup of the data on a regular basis to a folder on the SDCard (say every hour we sync
>the data).

>The only reason we kept the data in a shared folder was because I couldn’t get the same DEVICE_ID from the client when
>fetching them from the plugins. This can be a bug of my own doing, especially because the ContentProviders are set as
>exported, thus we should be able to query them, regardless of package. It would be great if you figure out how to get
>this working properly.

rkdarst, 2016-09-27:
So, as part of changes, I have been playing with DatabaseHelper.

I have tried to access Aware.getSetting() from inside of DatabaseHelper, to set the data location.  But I have the idea that it recurses and crashes, because settings are stored in a database managed by DatabaseHelper.  (this is just a hypothesis, I haven't had time to dig deep).  If that's the case, this is a bad strategy.  But then is there any way to do this?  Maybe in this case get the value of the setting from app storage or something?

Your idea of delete data on uninstall sounds more promising, but can we get uninstall hooks?  Somewhere I was reading and it wasn't possible for an application to hook its own installation.  Another idea: option in app, if server enables it someone can click to remove all data and uninstall?

This is somewhat important for us, since I need to know if we need to modify our proposals from "all data deleted when uninstalled", but it can wait until after Thursday...  (but thinking about it again, one study may want to us a plugin, which means that I can't even use my "put databases somewhere else" strategy.  So alternative thoughts would be important here!)

denzilferreira, 2016-09-29:
Hum... The main issue here is accessing the data from a different package. Before, I could not get the DEVICE ID to be the same as the one in the client for plugins, but to be honest I did not spend enough time to properly debug this and reverted to the sdcard shared folder approach.

So here's what I suggest:
- set the client to use it's own folder. ContentProviders that are exported should be readable from a different package anyway.
- set a plugin to use it's own folder. See what DEVICE_ID you get on your plugin's contentprovider. It should match the same as the client. If it does, problem solved.

rkdarst, 2017-07-12:
This exists enough for my needs now, so closing to help clean up.

closed 2017-07-12 by rkdarst