denzilferreira/aware-client #26: [wishlist] better ssl cert verification using QR code

rkdarst, 2016-06-18:
>> verification of pinned SSL cert hash via QR code parameter

> Can you explain to me why is this important? The user only gets the SSL after reading the QRCode. I guess you want to prevent a man-in-the-middle attack?

Yes, something like that.  Right now cert is fetched over non-ssl.  When making our first app, we had QR code embed `https://url?cert_sha256=hash&...`.  So, if the computer presenting QR code is secure, then you can establish the entire trust chain.  It's not perfect, for example it assumes the PKI works on the subject's computer if they are doing it themselves.  The disadvantage is that, for long studies, the cert couldn't be updated partway through.

What we could contribute:
- when getting config URL, strip off everything after `?`.  At least this could be done sooner, so that if options are added in the future, old clients won't break.
- The options after `?` can be querystring parsed and stored somewhere
- When getting server.crt (or ca.crt), if there is a hash given in qr code, verify it.

What do you think?  It is a relatively specialized idea that is only useful against active attacks.  Can you think of any better ways to do this?  (aside: we would implement server-side support in our server, but I'm not expecting this in the aware-server.  We actually have server-side support for this for another app).

Thanks,

- Richard

rkdarst, 2016-09-27:
See tetujin/aware-client-ios#4 - we have a strategy to pass certificates via the qrcodes.

rkdarst, 2016-10-03:
Done for Android, awaiting confirmation that it works.

denzilferreira, 2016-10-03:
Merged.

closed 2016-10-03 by denzilferreira