denzilferreira/aware-client #29: Condition for cleaning old data

rkdarst, 2016-06-18:
Hi,

In WebserviceHelper.java, bottom of function onHandleIntent, lines 301, I see (effectively) the following condition:

```
if (Integer.parseInt(Aware.getSetting(getApplicationContext(), Aware_Preferences.FREQUENCY_CLEAN_OLD_DATA)) == 4
    && highFrequencySensors.contains(DATABASE_TABLE)) {
```

Does this effectively mean that only the high-frequency tables are cleared (as listed right above)?

Could there be an option to clear all data once it has been sent to the webservice?  On a related note, in WebserviceHelper.java, is the POST response to `/insert` checked enough to ensure that data is only removed if the server verifies that it was successful?  Would having the server return something that is verified be useful?  (I guess with the aware-server, the latest row is retrieved from the server to provide this failsafe).

Thanks,

- Richard

denzilferreira, 2016-06-23:
Yes, only high-frequency tables are cleared immediately. The others are event based and don't get massive as the high-frequency sensors.

If you want to clear the data on the phone, you can send a broadcast to the device with clear data. That only removes the data from the phone, not the server :)

Added a check for the latest value on the server side before clearing locally, in case the upload gets interrupted for some reason.

Cheers,
Denzil;

rkdarst, 2016-06-23:
Hi,

Would you like to provide some advice on what we may do?

From what I see, we can see, timestamp can be used as an always-ascending key.  So, if the server responds with the largest timestamp it sees, then the client can delete all rows with a timestamp of less than that.  The client strictly sends in ascending order, and it's the server's job to atomically commit rows + last timestamp.  A client that doesn't know about this ignores the server response, which is backwards compatible.

Do you think this is a good system for our branch?  Any problems?  Any better ideas?

Thanks,

- Richard

denzilferreira, 2016-06-23:
I have already fixed this on the master branch. Check [here](https://github.com/denzilferreira/aware-client/blob/master/aware-core/src/main/java/com/aware/utils/WebserviceHelper.java#L308)

rkdarst, 2016-07-02:
Hi,

I've looked over this.  It's an interesting strategy.  This makes an extra HTTP request each time, even when there was no data to clean (e.g. if the sensor is off but the table still exists).

Also, from what I can tell the success status of HTTP calls are not checked.  The Http/Https classes return null if the response is not HTTP 200, but then this isn't checked.  This is normally OK because of the latest timestamp thing, but for uploads with multiple batches, there could be a problem if a first one fails but the later one doesn't.  So, we can make the uploading loop exit at the first null response.

I would still like to change the webservice API some, make clearing tables not dependent on broadcasts.  At least, if there is a mode to have auto-clearing, then we don't have to /create_table and /latest every time we sync (only do that when there are rows present), saving two HTTP requests for each infrequently updated table.

We can also use response code from the /insert for cleaning, instead of making a new request.  Overall, we can reduce network traffic which is important for battery life and reduce network usage and server load - all important things for scalability.

What do you think?  Any advice for us implementing these things and submitting pull requests?

denzilferreira, 2016-07-02:
Hi Richard,

I really appreciate another set of eyes over my code, and your insight to improve/fix it. I  encourage you to keep it up!

For the moment, I’m not modifying the client, so feel free to make pull requests with fixes on the master branch. The best way to contribute:
- Fork the client on the master’s branch
- Make changes on your fork
- Test your changes to make sure we are still getting the data sent to the server
- Make pull request (PR) when ready
- As soon as I approve the PR, Jenkins will take care of updating the client automatically and publishing it back to the Play Store.

Thanks!
Denzil

denzilferreira, 2016-07-09:
> I've looked over this. It's an interesting strategy. This makes an extra HTTP request each time, even when there was no data to clean (e.g. if the sensor is off but the table still exists).

If the sensor is off, it does not receive the sync request, so this is not a problem.

> Also, from what I can tell the success status of HTTP calls are not checked. The Http/Https classes return null if the response is not HTTP 200, but then this isn't checked. This is normally OK [...]

Yes, good catch. Fixed on the latest push.

> We can also use response code from the /insert for cleaning, instead of making a new request. Overall, we can reduce network traffic which is important for battery life and reduce network usage and server load - all important things for scalability.

This is an excellent suggestion. Modified and pushed on the latest.

closed 2016-07-09 by denzilferreira

rkdarst, 2016-07-10:
Hi,

> If the sensor is off, it does not receive the sync request, so this is not a problem.

OK, I see what you mean about this.  So, I should have said all enabled sensors, even if they have no new data.With my current config with minimal probes, there are 30 HTTP requests and four of these are /insert.  About half the extra ones are /create_table (which I ignore) and half are /latest (which requires DB access).  This is what I mean by a lot of queries.  I think I see why it is this way originally (must /create_table before /latest the first time and when changed, must /latest before /insert).  I think this shows what I am thinking about.

Also, on a somewhat related note: I have been discussing with some people to audit our setup (mainly about security and data protection) for getting approval for using AWARE and our server for projects.  One thing they strongly recommend is a write-only method of accepting data.  This would probably be limited as database user with permission only to write to certain tables.

So, I was planning this and realized that the /latest of AWARE requires the data ingestor to be able to read and modify state on the frontend.  I think this is at least somewhat related to the problem above.  (Of course, one could make specialized frontend for this, or a DB connection that can modify this state separate from the real data, but it seems better to have a simpler interface).

I was basically thinking of this now.  A mode (or branch) that does:
* (optionally) no /create_table requests
* no /latest requests
* Selects all data in the DB for sending
* After confirmation of saving data successfully (either timestamp/checksum return or HTTP 200), delete all the data that was just sent.  This is what prevents data from being sent over and over.  This deleting would have to be by max-sent-timestamp, so that 
* Before doing all of the above, check if there is new data.  If not, do nothing.
* This guarantees data is sent at-least-once, but not only-once.  only-once would have to be handled by some other means on the server (in my case, easy because of the always-ascending nature of the data).

What about config options named `web_only` and `web_stateless`?  `web_only` does the parts related to deleting local data, and `web_stateless` does things related to not fetching the latest timestamp from the server.

What do you think?

Thanks,

- Richard

rkdarst, 2016-07-10:
Hi,

Some more thoughts:
* variable names should have been `webservice_only` and `webservice_stateless`.  Is there something better than stateless?
* If `webservice_only` is true, this could also be used for putting the data into /sdcard/Android/com.aware.phone/... instead of /sdcard/, so that data is removed when uninstalled.  Would require some juggling when this is set later.
* I realized/think that right now there is a choice between cleaning high-frequency sensors immediately () and cleaning data daily/weekly/monthly, since they are controlled by the same variable `frequency_clean_old_data`, and "always" means "only high frequency".

Any thoughts on these?

Thanks,

- richard

denzilferreira, 2016-07-11:
> I should have said all enabled sensors, even if they have no new data.With my current config with minimal probes, there are 30 HTTP requests and four of these are /insert. About half the extra ones [...]

I've thought about this and I agree with you, we are doing way too many HTTP requests to enable syncing of the data. I'll need to think a little bit about how we can reduce this to hopefully 1 latest and 1 insert call (it could be possible to enable a persistent HTTP post) per sensor and each plugin.

> This would probably be limited as database user with permission only to write to certain tables.
Yes, you can definitely restrict the database user to write-only.

> Selects all data in the DB for sending
Unless you are thinking about syncing every 5-10 minutes, this will explode the memory on the phone/tablet if you are collecting high-frequency sensors at maximum speed. There is a limit on how many records you can hold in memory on the phone before you get a nasty overflow exception from the Cursor class.

> After confirmation of saving data successfully (either timestamp/checksum return or HTTP 200), delete all the data that was just sent. This is what prevents data from being sent over and over. This deleting would have to be by max-sent-timestamp, so that

We don't do this because of the AWARE Stream functionality. If we cleared the data, there would be no local data to show to the participants.

> Before doing all of the above, check if there is new data. If not, do nothing.
You risk getting yourself into a never-ending sync, especially with high-frequency sensors active. Essentially, they do not stop collecting data while you upload data, so you will always have data to send to the server...

rkdarst, 2016-07-11:
Hi,

The "web-only" mode would not work with AWARE stream or things like that.  (for some ethics permissions I know of, not storing any data on the phone long-term is a requirement.)  In the simplest case, this means that it just doesn't do anything.  I know that this is outside the original vision of AWARE, but things don't have to be perfect.  Our studies either have people able to see their data on the server only, or they can't see their data at all (if that's what is required).

> You risk getting yourself into a never-ending sync, especially with high-frequency sensors active. Essentially, they do not stop collecting data while you upload data, so you will always have data to send to the server...

Hm... well now the sync eventually stops.  I was thinking something like: select max timestamp, send everything leq that timestamp, delete everything leq that timestamp.  Anything new synced next round.

>> variable names should have been webservice_only and webservice_stateless. Is there something better than stateless?
> Not sure if I follow correctly. How exactly would these variables affect the client and synching?

* `webservice_only`  (human interpretation: don't store data on device, there isn't useful data stored without webservice)
  * deletes data after being sent
  * visualizing or using data on device is not possible
  * Not in UI, only set via server configuration
* `webservice_stateless` (human interpretation: server is dumb and doesn't know anything about particular clients, in particular doesn't store timestamps for /latest)
  * doesn't do /create_table
  * doesn't do /latest before sending - sends all data each time
  * deletes data after sending (using procedure above)
  * data may be sent more than once, user beware
  * not in UI, only in server configuration
* `use_app_storage`  (I just thought of this, not sure if it makes sense.  human interpretation: put data in app storage so that it is automatically removed)
  * plugins don't work
  * All data cleaned up when uninstalled
  * Not in UI, only by server configuration

All fall back to current behavior if unset.

For the /sdcard/ storage, I just though of something else: ethics/security people may not like the idea of one app collecting data, and it possibly being available for other apps who haven't requested permissions to collect that data.  I hadn't even thought about this before... for now I'll ignore it.

> Problem with putting the data there is, plugins can't read the data from another package, in this case, the client. The databases need to be located in a world-readable folder.

Ohh... very good point.  So in this case there can't be plugins (OK in at least some of our cases...).  My research shows there's no supported way for an app to get its own uninstallation notification.  So maybe our branch will have to not support plugins, or else user cleans up themselves.

> We can get about 2GB of accelerometer in a week if we log at full speed. You would be surprised how many times this is a problem.

In our spring study, we had some phones where accelerometer logging without threshold made them completely unusable for the subject.  So... yes, we know.  We ended up going with the threshold, which made it work for all devices.

Thanks for the ideas!

- Richard