denzilferreira/aware-client #30: [wishlist] sensor thresholds and other options

rkdarst, 2016-06-18:
Hi,

Previously, we were using Purple Robot.  It had configuration options for some sensors, like accelerometer, which would only generate a new data row if the new value was more than some threshold away from the old value.  For example, this was useful on accelerometer which would be  generating 1GB of data per day without the threshold, but with it it was small.  The data excluded was mostly zeros while the device was at rest.  Several other probes had the same idea.  The threshold was configurable as an option per-sensor, and there was usually a "0" option that would produce full data.  This allowed higher frequency sampling when data was interesting while lower when it was not.

This was important since some phones had limited storage, and storing too much data severely impacted them.  Also uploading so much data can have an effect on the subjects network usage, which we would then need to consider.

(on another note, where were options such as "high frequency sample for N seconds, every M minutes).

I could have my team study and make a list of these types of options that would be useful.  We could even try to implement them.  Do you think this is something that would be useful to you?

Thanks,

- Richard

denzilferreira, 2016-06-23:
Yes, this is of interest. We were also considering setting a sampling schedule.

rkdarst, 2016-07-27:
I have just outlined a draft plan for sensor thresholds in our branch.  You can see and comment on the issues here: cxaalto/aware-client#2

What would you consider the minimum requirements for merging into upstream?  What are the different feature-levels we should target for PRs ?
* Setting a different default changes upstream behavior, which maybe you don't want to do (though I would think it makes sense)
* We could make a patch with the configuration option, but default to threshold of zero (no behavior change).  Then our branch has another patch changing the default value.
* We would need help making sure everything fits together well.
* If you want to do this yourself, let us know soon!

denzilferreira, 2016-07-31:
Took a look at this. threshold_ variables seems good to me and can be a setting of each sensor. One feature at a time per push, so we can test them.

rkdarst, 2016-09-10:
Hi,

I have a proposed patch here, would you mind providing feedback on its completeness and suitability in the big picture of Aware?
  https://github.com/CxAalto/aware-client/commit/0a9b429b23f6713d0f852e361515a990b62b774b

It's not ready for pulling yet, I may do more fixup and if it looks good, then do the same for the other sensors.  Also this commit hash is likely to disappear due to rebasing and squashing eventually.

Thanks,
- Richard

denzilferreira, 2016-09-11:
Looks good to me! It's a welcome addition!

Denzil;

CxAalto@ 0a9b429 [https://github.com/CxAalto/aware-client/commit/0a9b429b23f6713d0f852e361515a990b62b774b]

rkdarst, 2016-09-11:
Right now Aware is set with `maxReportLatencyUs` (microsecond latency, amount which we are willing to wait to get a value) to zero, which means we want to be as real-time as possible.  Perhaps we could reduce power consumption by specifying something larger here.  Does Aware need zero latency?  Could we set this to something larger, and if so do you have a suggestion?

(SensorEvent has a "timestamp" attribute which can be used to get the time of the actual event, which I may change it to use anyway for higher accuracy).

denzilferreira, 2016-09-11:
@rkdarst: Add the threshold also to aware-core module (xml/aware_preferences.xml and Aware_Preferences list of variables), as that's what is bundled as a library.

Where is this maxreportLatencyUs? I'm guessing you mean the HAL FIFO buffer that I recently added to all high-frequency sensors?

rkdarst, 2016-09-11:
Re: latency, I now see this that adjusts it: FIFO_SIZE = mAccelerometer.getFifoReservedEventCount();`

This is the function we use the register the listener:
https://developer.android.com/reference/android/hardware/SensorManager.html#registerListener%28android.hardware.SensorEventListener,%20android.hardware.Sensor,%20int,%20int,%20android.os.Handler%29
fourth argument is: maxReportLatencyUs: "int: Maximum time in microseconds that events can be delayed before being reported to the application. A large value allows reducing the power consumption associated with the sensor."

I thought it was zero, but now I see that `FIFO_SIZE` is changed to `getFifoReservedEventCount`.  However, `getFifoReservedEventCount` doesn't seem to be in units of time, while `maxReportLatencyUs` seems to be microseconds.  So now I'm confused to what it does.

Also, I see that the `registerListener` was added in API level 19.  Does this mean that Aware requires android 4.4 or higher?  I see build.gradle says `minSdkVersion 10`, but I'm not sure how this works.

Thanks for the hint on aware-core aware_preferences.xml.  I'll add the parts there.

denzilferreira, 2016-09-11:
Hum... Good catch... I didn't notice the API level. We'll need to make this backwards compatible with API checks in code. I'd like to keep the 2.3.3+ compatibility.

The reserved FIFO events is in microseconds, as far as I could test on my device. I get values as 3000 for accelerometer.

rkdarst, 2016-09-11:
I've checked and my interpretation is that FIFO_SIZE is the number of events that hardware can store, before having to wake up the OS.  And the microsecond count is maximum latency our application will accept.  So, in your case, hardware can hold up to 3000 events (=10 min at 200ms frequency) before having to wake up the OS, but we are telling it wake us up within 3ms of generating each event.  If we don't need low latency, I would say we could make it 60 (or 600?) seconds, or `FREQUENCY*FIFO_SIZE` to optimally match the hardware if we really don't care about prompt updates.

By the way, should we set a default threshold?  Right now it is zero for everything.  I'm assuming it is hard to pick something that is useful for all devices and all sensors without at least checking each sensor individually.  But setting it to something low, would make data much more efficient for most sensors while the phone is not being manipulated (e.g. dark at night).  I could make investigating this a project for someone.

PR coming soon.

rkdarst, 2016-09-15:
Thanks for getting the patch on the thresholds!

Have you thought about `FIFO_SIZE` and `maxreportLatencyUs`?

However, if we did that, we would need to change the way we get the timestamp, from  `System.currentTimeMillis()` to using the `Event.timestamp` attribute.  However, if you search "android SensorEvent timestamp", you see there's a major problem.  This can vary across hardware, with both scaling (ms vs µs vs ns) and offset (unix epoch vs boot time) changing!  This becomes a major issue and I haven't seen anyone with a good solution yet.  Example (but there are many more): https://code.google.com/p/android/issues/detail?id=56561#c3

What do you think? 

Thanks,
- Richard

denzilferreira, 2016-09-16:
Thanks for looking into this. 

Since sensing of time series really depends of the device-specific timestamp, I believe that this is a major issue of manufacturer fragmentation which is beyond our control but it's not much of an issue unless you are doing cross device temporal analysis.

Perhaps adding an API check for the registration of the sensors for the support of the FIFO/maxreportlatencyus in API 19+ and record both timestamps, one from the System.currentTimeMillis() and the other one from the event.timestamp. The System.currentTimeMillis() would allow for cross-device temporal analysis since the timestamp is NTP synched.

rkdarst, 2016-09-17:
In my case, we really need real times since we are comparing to real-world events and other devices.  One idea I saw someone suggest was to get several events a second apart, and use that to reverse-engineer what the timestamps mean.  This is still really annoying to do though...

With your server and framework, are there problems if timestamps are not unixtime ms?

Then there's the issue of 2.3 compatibility.  Is the right way to do this to use `android.os.Build.VERSION.SDK_INT` and then a normal `if` conditional?

Thanks,
- Richard

rkdarst, 2016-09-18:
I'm going to close this issue and reopen one about the new issues which have come up.

See #46

closed 2016-09-18 by rkdarst