denzilferreira/aware-client #31: [bug] Configuration lost when upgraded through play store

rkdarst, 2016-07-02:
Hi,

On two devices now, we have lost configuration when AWARE has been updated through Google Play.  There haven't been enough upgrades to repeat this several times to get more info.

The exact symptom is that AWARE is re-opened, and then you see in the UI that configuration options are not set (seen most clearly in the study configuration).

Any thoughts?  We will update this issue when we observe more.

Thanks,
richard

denzilferreira, 2016-07-02:
Hum, I've updated countless of times and the previous settings always prevailed. I'll try to take a closer look at this. Let me know what you find.

rkdarst, 2016-08-27:
I think this is no longer the case.  I don't know why it was.  I think it is fixed for others now too.  No idea what caused it...

closed 2016-08-27 by rkdarst