denzilferreira/aware-client #32: [bug] force-stopping AWARE disables it as accessibility service

rkdarst, 2016-07-02:
Hi,

I noticed that if I force-stop AWARE (though Android settings), then it is no longer enabled as an accessibility service.  This is quite an issue, since from what I understand then it won't remain running in the background.  There is no notice of this once you re-open the app.  This is a big problem for us, since as studies get longer, the chances of this happing to some users increases.

Maybe one work-around is to make it produce a notification when the app is opened and services are not turned on?  Beyond that, I guess this is just the way Android works.  (As an aside, in the long-term, we would be interested in a mode that had a persistent notification and wasn't dependent on accessibility services.  We would develop this if we needed it.)

Device: Nexus 5 running latest Google Android.

Any thoughts?

Thanks,
Richard

denzilferreira, 2016-07-02:
Hi Richard,

Yes, the accessibility services can be a bit problematic. There is scarce documentation for them, and what you find is most often outdated. As far as I tested, if you explicitly force stop it, Android by default turns it off. It stays off until the user explicitly turns it back on. It's like that by Android policy and design.

Since you are force-stopping it (this does not happen in normal conditions), there may be a consistency check missing. Note that we are keeping an internal state variable for the accessibility service being active or not (STATUS_AWARE_ACCESSIBILITY) which takes precedence over directly querying the AccessibilityManager for the active services (simply because it is buggy and sometimes returns null when it should return true when AWARE is indeed active still on the UI). To fix this, try setting the STATUS_AWARE_ACCESSIBILITY to false when the accessibility service is force-stopped (not sure if there is a way to detect this).

Another suggestion to fix this:
- Make AWARE able to receive the on boot completed broadcast by adding the permission to the manifest on the aware-core.
- On boot completed, check if the accessibility service is active. If not, notify the user.

We used to have a persistent notification many cycles ago, and unfortunately, researchers/users did not like that because it was a constant reminder to the participants that they were on a study, thus potentially biasing their "normal" behavior. So we drop it. Since we need the accessibility services to be able to log the app usage and keyboard, we did not look back.

denzilferreira, 2016-07-09:
Fixed in latest push.

closed 2016-07-09 by denzilferreira