denzilferreira/aware-client #33: deterministic upload times, and unreliable uploads

rkdarst, 2016-07-02:
Hi,

This is continued from a private email thread.  I am putting it here for openness and recordkeeping.

I was trying to see why AWARE was not reliably uploading data.  I eventually saw that it uploaded at fixed intervals of the hour.  For example, if the upload frequency was 15 (min), it would upload on the [:00, :15, :30] of the hour.  I can see the following possible issues:

- Due to a bug, it doesn't upload at the :00 of the hour
- The uploads are deterministic and the same for all clients, which is bad for scalability!
- I haven't tested yet, but there may be issues with frequencies longer than 60 minutes.

These can be solved with some code changes, and then adding a random offset from [0, freq-1].  Code link: https://github.com/denzilferreira/aware-client/blob/master/aware-core/src/main/java/com/aware/Aware.java#L399  (second copy right below)

However, even with the above, Android returns control to the application at non-deterministic times.  This overall makes sense and is just how it works.  However, the scheduler only runs the upload process if the process runs exactly on the minute.  When the application is not getting many resources (as it normally shouldn't), this is not often enough and it doesn't upload reliably.

So... I would think that the best answer is to understand in incorporate this non-determinism.  For something like uploading, we shouldn't try to force it to upload at fixed offsets, but have a scheduler mode that can say "no more often than every X minutes, do this task".  Then, uploading can happen anytime the scheduler gets called, if it hasn't already in the last X minutes.  This also nicely solves the fixed-offset problem, and then we don't have to worry about the schedule runner being non-deterministic.  We can then run the schedule runner less often, which may encourage Android to give it a higher priority...

What do you think?  We can at least fix the schedule to be non-deterministic.  Do you have any suggestions for adding a frequency-based scheduler, too?

Thanks,
Richard

denzilferreira, 2016-07-02:
Hi Richard,

Yes, this sounds like a good fix for the by exact minute issue. Maybe replace the logic of the type minute to an interval type of Scheduler. It used to be like this before, using an Alarm on the AlarmManager, but changing it to a Scheduler seems indeed a regression on functionality as it is implemented now.

Feel free to modify the Scheduler to support the interval type and remove the minute functionality (since it's broken) and then modify the Aware class to register this new schedule and react to it properly. This change should allow us to:
- schedule X (activity, service, broadcast) to run Y times with at least a gap of Z minutes every T minutes
- schedule X randomly between X and Y hours of the day, up to Z times, with at least T minutes apart

rkdarst, 2016-07-03:
This is the trace of scheduling stuff, to help my coworker who may look at this (I said I would post this here, everyone else please ignore).

* aware-core/src/main/java/com/aware/Aware.java lines 398
  * Doing the minute-addition to the schedule (addMinute).  Replace with frequency-based mode.
* aware-core/src/main/java/com/aware/utils/Scheduler.java
  * scheduler that does everything.  I guess addFrequency, then is_trigger handles running it.  It already calculates the last_triggered time.

denzilferreira, 2016-07-09:
Fixed in the latest push.

closed 2016-07-09 by denzilferreira

rkdarst, 2016-07-10:
I looked over it.  Looks good, thanks!

p.s. I'm sorry I/we haven't been able to contribute code yet...

rkdarst, 2016-07-11:
I now hear from two people (me+one other) that this works.  Also, it's nice that it automatically removes the notification now.  Very useful.

Thanks,

- Richard