denzilferreira/aware-client #38: When phone starts, it requests bluetooth be turned on even if it is already on

rkdarst, 2016-08-21:
Hi,

When I turn on my phone, it makes a notification saying 'an application want to turn on bluetooth, accept or deny'.  It seems this is produced by the OS, not explicitly app-rendered (but I could be wrong here).  When this happens, bluetooth is already turned on.

I don't remember this happening months ago, but I can't say for sure when it started.  I think I do have the bluetooth sensor enabled.

Also, this is a separate issue, but the dialog doesn't say what app is requesting it or why.  This is probably because it is a system dialog and it can't say anything else.  Still, I think it's worth asking how the user experience could be improved, especially for long studies with only limited connection, so that people forget what is going on.

Thanks,
- Richard

denzilferreira, 2016-08-24:
Hum... are you sure this is caused by AWARE? This seems to be an issue with Android itself.

rkdarst, 2016-08-24:
I just did some tests.  Nexus 5, android 6.0.1 (latest as of now).  It very well could be the android system, so I did some tests.

For reference: the exact dialog is "An app wants to turn Bluetooth ON for this device.  \n\n DENY      ALLOW"

Case 1: System BT=on Aware BT sensor=on
* Power down and reboot
* The dialog appears

Case 2: System BT=on, Aware BT sensor=off
* Power down and reboot
* The dialog does not appear

Case 3: repeat of case 1: same behavior

Case 4: Turn on and off system BT, then repeat case 1: same behavior

Case 5: System BT=off, Aware BT sensor=on
* Power down and reboot.
* Dialog comes up.  (The system BT icon is on in the background already).  This is same behavior as normal.
* Click "Deny".
* System BT is on!

Case 6: System BT=off, Aware BT sensor=on
* Repeat case 5, but click "allow".
* Same behavior as case 5.

Case 7: from within Aware app
* System BT=off
* Go to Aware BT sensor configuration, and turn it off and on again.
* The dialog appears
* If you click allow, system BT is turned on, if you click deny, it is not turned on.

Just to be safe I uninstalled Purple Robot, it's the only thing I can think of that might involve bluetooth, but I can't discount others.  (However, there is clearly a relation to Aware BT sensor being on or off).

Thanks,
- Richard

I'm not aware enough of android internals to know just what is going on here...

denzilferreira, 2016-08-24:
Thanks for testing. Indeed looks a strange behaviour. I'll look into it.

rkdarst, 2016-08-24:
Several more tests:

* Nexus 5X:
  * Basic behaivor remains
  * If system BT=off, Aware BT sensor=on  and you reboot:  The prompt appears twice this time.  She clicked "deny" twice.  At first BT appeared off, but then it turned on!
  * Otherwise pretty similar to the cases on my device.  BT gets automatically turned on even if you deny.  This only happens if Aware BT sensor=on.

* Samsung Galaxy Mega (AWARE version 583, an older version):
  * Basically the same behavior, sometimes prompts twice.

Thanks,
- Richard

denzilferreira, 2016-08-24:
Found the issue, will be soon fixed.

commit 83d15f90 (2016-08-24): - Fixed #38: Custom notification to enable deactivated bluetooth when scanning.
- Improved UI consistency when installing AWARE for the first time (updates the interface with default preference values without restarting activity).
- WIP: dynamic permissions per sensor (similar to plugins/apps), instead of asking all the permissions on first-run.
closed 2016-08-24 by denzilferreira