denzilferreira/aware-client #40: Settings to control hashing

rkdarst, 2016-08-27:
Hi,

I'd like to propose some settings that can adjust hashing of sensitive information.  In summary:
* An option to set the hash algorithm
* An option to salt the values hashed
* Abstracting out the hashing to another function
* Option to specify different hash functions for different types of data

In detail:

Option to set the hash algorithm + separate function
* Right now, in android and iOS, I see sha1 hard-coded in different places.  We only need to worry about reversals, but the differences in brute forcing things like phone numbers for them all is not that much.
* The abstracting out the hashing to another function is needed for other things, so may as well make it configurable.
* I would suggest options be at least md5, sha1, sha256.  sha1 should be default for backwards compatibility.  Maybe md5 could be excluded, but could be useful if researchers need to compare to other data sources.
* Controlled by variable `hash_function` which is the string of hash function names.

Option to salt the values hashed
* Add a new variable, `hash_salt`.  It is a string defaulting to the null string `""`.
* This really should be done, there are websites which can reverse hashes of phone numbers, as students have found out.  Salt doesn't protect against brute force, but at least our hashes should not be in rainbow tables!
* `hashed_value = H(value || salt_string)`.  This is string concatenation.  I don't think any of the extra security of HMAC is necessary here, since the salt_string is public.

Option to specify hash functions for different types of data
* This means that, for example, you could specify a different hash function for phone numbers as compared to other things.
* The motivation for this is that we are doing a study, and we need to connect app-recorded phone numbers with numbers collected by surveys.  Asking for whole phone numbers is too invasive, so we settled on collecting only the last six digits.  We need to be able to uniformly hash this
* I would expect implementation would be something like adding a variable `hash_function_phone`, and have some way to use this if it is set, otherwise use the default `hash_function`.  People with special needs could define a custom `hash_function` that has their policy.

By the way, Purple Robot use the function `android.telephony.PhoneNumberUtils.formatNumber` to standardize the number formatting.  This was quite useful and important.  We should consider this, but for the non-deprecated `formatNumber` functions, we need to specify a country code and I'm not sure what to use there.

What do you think?
- Richard

denzilferreira, 2016-09-04:
Hi Richard,

Good points! It's been a while since we have gone through the privacy measures on the framework. At the time, SHA1 was the standard, but yes, we should move on to SHA256 or even SHA512.

As far as I know, it's impossible to get one's phone number programmatically on Android, except for rooted devices, but not on general population devices. Why not use the participants' email? We do have the Google Auth plugin that you could use to match these without being so invasive as you put it.

There are a few covenants regarding these changes:
- if you have already collected data with a specific participant and you recruit them again, if the hash changed, you won't know it's the same participant. This is important for longitudinal studies. Not a problem if we always pick the same hash algorithm.
- the non-existing uniform hash of phone numbers: this is actually intentional. Like this, you would get always the same hash for the same string on your device, but maybe not the same if on another device (if the number was input on the contact list differently). This blocks cross-device hash matching, as in, the same contacts are present in both devices, hence a common contact. This could have significant impact on social-engineering attacks.

rkdarst, 2016-09-11:
Hi,

Regarding getting phone numbers: we do not need phone numbers of participants, we know who the participants are.  We need to know something about who the subjects are calling.  For that, we have the remote phone number.

As for non-uniform hash: it seems there are better ways to prevent linking.  They could be hashed with a random salt that never leaves the device, and then there is actual security.  Or or, say, the device ID if you want to be a bit less secure.  (Even using device_id and if it was public, hash(data||device_id), there is no way to correlate things without brute force reversing).  (Another random thought: are incoming and outcoming numbers formatted the same way?  Does incoming use )

Still, in our use cases we need to be able to correlate remote contacts, so even if intentional we need to improve things.  I'm well aware of re-creating the social network, that is related to our research.

[theoretical, for reference: As for hash functions, I have tried to see what the situation with SHA1 is as it applies to us.  There is a risk of collision attacks, but this actually doesn't apply to us (it applies to things like digital signatures).  We are concerned with first preimage attacks, which is much harder and still theoretical even for md5.  Plus, even without standardizing the formats, there are so few bits of entropy that it could just be directly brute forced.  So debating hash functions is probably mostly theoretical.]

Given the above, we can mostly chose what we want.  sha256 seems like a reasonable option to provide.

As for changing the hash function during studies, that is indeed a problem.  That is why it is server-configurable (and the research needs to beware), but it defaults to sha1 so the default case is preserving compatibility.

As in all my changes, it would default to the existing behavior, and the options do not even need to be presented in the client interface or upstream Aware server, so your studies are not affected.  I would very much like as much of our changes to be accepted upstream as possible, to minimize the change of a hard fork.  Given this, would you like a PR even if you won't use the changes?

- Richard

denzilferreira, 2016-09-11:
Yes, PR so I can see the changes and merge upstream.

Denzil;

rkdarst, 2016-09-11:
I just made the PR.  The basic things work for me, and hashing is as expected.  Let me know if there are stylistic or java-related things I don't know about.

It is configurable, with all defaults to the existing settings.  Configuration is only be server, users don't see anything (if not using webservice, these are probably not that useful).  It is somewhat flexible, but not as generic as it could be.  I took the idea that there's no reason to over-engineer now, but at least if improvements are needed later, they are only needed in two functions.  Let me know if there are an particular use cases you are interested in now.

Also, I noticed that my phone returns the number in as standard form: no spaces or anything else, even though in contact list it is different.  In this case, the goal to obfuscate things doesn't work.  I added a option for the salts hash_salt="device_id", which will use the device_id as a salt for the hash.  This can be either applied only to phone numbers, or to all hashes.  I'm not entirely happy using device_id for this, but if you can think of a better idea let me know.

Thanks,
- Richard

rkdarst, 2016-09-17:
Now merged, so closing.  If someone has more requests for hashing, reply here.

closed 2016-09-17 by rkdarst