denzilferreira/aware-client #41: [question] is MQTT the only way to send ESMs, and other ESM questions?

rkdarst, 2016-08-27:
Hi,

I was investigating ESMs, to see if I can get it to work for our purposes.  I understand some parts of it, but just to make sure I wanted to ask some questions to make sure I get it right... I'm sorry if this is a lot.

Is MQTT the only way to send ESM questions to a device?  Is there any way to send them as part of the config file?  And then, is each question asked immediately: is there any way to send a bunch of questions and have them asked at specified times throughout the day?

I get the idea that MQTT persistence would ensure that a message is received even if the device is off/disconnected at that time.  Is that correct?

In the iOS client, I see that a lot of MQTT code is commented out.  Is it currently working?  If not, and if there's no other way to do ESM, can they be done in iOS ?

Thanks,
- Richard

denzilferreira, 2016-08-31:
@tetujin should answer the iOS related question.

Regarding MQTT on Android, yes this is the only way for now that you can send an esm request remotely. There is some work in progress to add support to predetermined esm questionnaires to a study, but it's not ready yet.

You can however write a plugin that contains a scheduler which all it does is trigger a question according to your needs and scheduled times :)

denzilferreira, 2016-09-01:
I've fixed today the MQTT functionality by updating Mosquitto to the latest. The MQTT persistence now works properly, delivering non-routable messages when the user connects back online.

tetujin, 2016-09-02:
Hi @rkdarst

I'm @tetujin.  On the iOS client, as you pointed out, MQTT related code is commented out now because it isn't working well in the background. I'm still developing it.

Also, about ESM, you can implement ESM to AWARE iOS directly. Please check the attached tutorial for ESM on iOS client.
http://www.awareframework.com/development-esms-on-aware-client-ios/

Another solution is to use WebESM plugin. (I'm sorry, I have not made the document yet.)
The plugin can download ESM configuration file(JSON format) from any server and make ESMs with notifications based on the configuration file.
https://github.com/tetujin/aware-client-ios/tree/master/AWARE/AWAREFramework/Plugins/WebESM

However, now we don't have a solution to send any ESM anytime. 
To send notifications (as a reminder) from a server side, I'm using push notification function.

Thanks,
@tetujin (Yuuki)

rkdarst, 2016-09-18:
Thanks for all answers... for now we are going with our former solution, but sometime I'll come back to this.

closed 2016-09-18 by rkdarst