denzilferreira/aware-client #42: Exceptions when joining a study

JulioV, 2016-08-30:
After I scan the QR of my Aware server I get two exceptions, one related to the MQTT URL and the other to an unclosed cursor in the Scheduler:

```
08-30 11:41:25.680 13378-13378/com.aware.phone W/ContextImpl: Failed to ensure directory: /storage/sdcard1/Android/data/com.aware.phone/files
08-30 11:41:25.680 13378-13378/com.aware.phone E/AWARE: com.android.org.conscrypt.OpenSSLX509CertificateFactory$ParsingException: inStream == null
08-30 11:41:25.877 13378-13378/com.aware.phone D/AndroidRuntime: Shutting down VM
                                                                 
                                                                 
                                                                 --------- beginning of crash
08-30 11:41:25.882 13378-13378/com.aware.phone E/AndroidRuntime: FATAL EXCEPTION: main
                                                                 Process: com.aware.phone, PID: 13378
                                                                 java.lang.RuntimeException: Unable to start service com.aware.Mqtt@2c4c5b44 with Intent { cmp=com.aware.phone/com.aware.Mqtt }: java.lang.IllegalArgumentException: ssl://https://mystudy.co.uk:8883
                                                                     at android.app.ActivityThread.handleServiceArgs(ActivityThread.java:2975)
                                                                     at android.app.ActivityThread.access$2100(ActivityThread.java:156)
                                                                     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1449)
                                                                     at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                     at android.os.Looper.loop(Looper.java:211)
                                                                     at android.app.ActivityThread.main(ActivityThread.java:5373)
                                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                                     at java.lang.reflect.Method.invoke(Method.java:372)
                                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1020)
                                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:815)
                                                                  Caused by: java.lang.IllegalArgumentException: ssl://https://mystudy.co.uk:8883
                                                                     at org.eclipse.paho.client.mqttv3.MqttConnectOptions.validateURI(MqttConnectOptions.java:458)
                                                                     at org.eclipse.paho.client.mqttv3.MqttAsyncClient.<init>(MqttAsyncClient.java:273)
                                                                     at org.eclipse.paho.client.mqttv3.MqttAsyncClient.<init>(MqttAsyncClient.java:167)
                                                                     at org.eclipse.paho.client.mqttv3.MqttClient.<init>(MqttClient.java:224)
                                                                     at com.aware.Mqtt.initializeMQTT(Mqtt.java:414)
                                                                     at com.aware.Mqtt.onStartCommand(Mqtt.java:352)
                                                                     at android.app.ActivityThread.handleServiceArgs(ActivityThread.java:2958)
                                                                     at android.app.ActivityThread.access$2100(ActivityThread.java:156) 
                                                                     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1449) 
                                                                     at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                                     at android.os.Looper.loop(Looper.java:211) 
                                                                     at android.app.ActivityThread.main(ActivityThread.java:5373) 
                                                                     at java.lang.reflect.Method.invoke(Native Method) 
                                                                     at java.lang.reflect.Method.invoke(Method.java:372) 
                                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1020) 
                                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:815) 
08-30 11:41:29.865 13378-13390/com.aware.phone E/Cursor: Finalizing a Cursor that has not been deactivated or closed. database = /storage/emulated/0/AWARE/scheduler.db, table = scheduler, query = SELECT schedule, timestamp, device_id, _id, schedule_id, package_name, last_triggered FROM scheduler
                                                         net.sqlcipher.database.DatabaseObjectNotClosedException: Application did not close the cursor or database object that was opened here
                                                             at net.sqlcipher.database.SQLiteCursor.<init>(SQLiteCursor.java:217)
                                                             at net.sqlcipher.database.SQLiteDirectCursorDriver.query(SQLiteDirectCursorDriver.java:53)
                                                             at net.sqlcipher.database.SQLiteDatabase.rawQueryWithFactory(SQLiteDatabase.java:1772)
                                                             at net.sqlcipher.database.SQLiteQueryBuilder.query(SQLiteQueryBuilder.java:330)
                                                             at net.sqlcipher.database.SQLiteQueryBuilder.query(SQLiteQueryBuilder.java:280)
                                                             at com.aware.providers.Scheduler_Provider.query(Scheduler_Provider.java:213)
                                                             at android.content.ContentProvider.query(ContentProvider.java:1000)
                                                             at android.content.ContentProvider$Transport.query(ContentProvider.java:214)
                                                             at android.content.ContentResolver.query(ContentResolver.java:482)
                                                             at android.content.ContentResolver.query(ContentResolver.java:426)
                                                             at com.aware.utils.Scheduler.getSchedule(Scheduler.java:150)
                                                             at com.aware.Aware.onStartCommand(Aware.java:415)
                                                             at android.app.ActivityThread.handleServiceArgs(ActivityThread.java:2958)
                                                             at android.app.ActivityThread.access$2100(ActivityThread.java:156)
                                                             at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1449)
                                                             at android.os.Handler.dispatchMessage(Handler.java:102)
                                                             at android.os.Looper.loop(Looper.java:211)
                                                             at android.app.ActivityThread.main(ActivityThread.java:5373)
                                                             at java.lang.reflect.Method.invoke(Native Method)
                                                             at java.lang.reflect.Method.invoke(Method.java:372)
                                                             at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1020)
                                                             at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:815)
```

After a few minutes Aware tried to resume itself but I got the same exception with the MQTT URL and a new one with an unclosed cursor in Ligth.

```
08-30 11:46:29.338 6054-6054/com.aware.phone E/AWARE: com.android.org.conscrypt.OpenSSLX509CertificateFactory$ParsingException: inStream == null
[... cut ...]
                                                           at android.app.ActivityThread.main(ActivityThread.java:5373)
                                                           at java.lang.reflect.Method.invoke(Native Method)
                                                           at java.lang.reflect.Method.invoke(Method.java:372)
                                                           at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1020)
                                                           at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:815)
```

denzilferreira, 2016-08-31:
ssl://https://mystudy.co.uk:8883

On the MQTT server URL, use mystudy.co.uk, not https://mystudy.co.uk. That should fix it.

JulioV, 2016-08-31:
That was it, I changed $config['mqtt_hostname'].

Thanks!

closed 2016-09-01 by denzilferreira

sukhwinder-kaur, 2018-07-20:
HI . I am using this type url 
mqtt://mqtt.go.th:1883 but with this i can't get make connection between my android app and mqtt server.
But when i am trying it with this tcp://mqtt.go.th:1883 it make connection . what happened with mqtt:// ? I need this type url connection. please give me solution .
Thank you.

denzilferreira, 2018-07-20:
Not supported by Mosquitto, the server we use in our server. You can nonetheless edit the client code to add the support for your needs.

Denzil;