denzilferreira/aware-client #43: [BUG] off-by-one error in webservice uploader

rkdarst, 2016-09-04:
Hi,

I see this error message in adb log.  The "sync accelerometer OK" means that hte first-round of syncing succeeded, but it is still trying.  Thus, the error comes during the second round (syncing records [10000,19999])
```
09-04 02:07:48.244  9735 10825 D AWARE   : Sync accelerometer OK
09-04 02:07:48.249  9735 10825 W System.err: org.json.JSONException: Index 10000 out of range [0..10000)
09-04 02:07:48.257  9735 10825 W System.err:    at org.json.JSONArray.get(JSONArray.java:293)
09-04 02:07:48.257  9735 10825 W System.err:    at org.json.JSONArray.getJSONObject(JSONArray.java:521)
09-04 02:07:48.257  9735 10825 W System.err:    at com.aware.utils.WebserviceHelper.onHandleIntent(WebserviceHelper.java:348)
09-04 02:07:48.257  9735 10825 W System.err:    at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:66)
09-04 02:07:48.257  9735 10825 W System.err:    at android.os.Handler.dispatchMessage(Handler.java:102)
09-04 02:07:48.257  9735 10825 W System.err:    at android.os.Looper.loop(Looper.java:148)
09-04 02:07:48.257  9735 10825 W System.err:    at android.os.HandlerThread.run(HandlerThread.java:61)
```

It is trying to do the steps needed to delete the old values so that it can upload the rest.  However, it fails on this line:
```
 long last = rows.getJSONObject(rows.length()).getLong("timestamp");
```
The error message says it all:
```
org.json.JSONException: Index 10000 out of range [0..10000)
```
If I understand this right, it should be `rows.length()-1` instead of `rows.length()`, since indexes go from [0...9999].

The effect is that on my device, syncing is about one day behind, each 15 minutes it gets about 5000 new points and syncs 10000, and I guess due to circumstances it falls behind sometimes.   If webservice frequency was lower, data would accumulate forever.  Also data isn't being deleted.  This behavior has been seen on two phones.

While fixing this, it would be nice for "always clean old data" to clean old data from all tables, not just high frequency ones (line 346), as I have mentioned before.

Also, at one point you mentioned something suggesting that the web upload API was improved, but I see the same large number of requests as before (for every table, for every upload attempt, /create_table and /latest every time regardless of what data their is).  Looking back at that, maybe it just meant that you wouldn't /create_table and /latest or *each* 10000 row chunk.  Should I expect any other behavior?  And are there any more improvement expected?

Thank you,
- Richard

denzilferreira, 2016-09-04:
Hi Richard,

> If I understand this right, it should be rows.length()-1 instead of rows.length(), since indexes go from [0...9999].

Good catch! Thanks for reporting this. I've fixed it on the latest push.

> While fixing this, it would be nice for "always clean old data" to clean old data from all tables, not just high frequency ones (line 346), as I have mentioned before.

Done, but not all tables are wiped, otherwise we get duplicates on the database with our current webservices. The aware_device and sensor_x tables, which contain the existing device's device ID and a snapshot of the sensors' capabilities is kept. They only contain 1 row of data, so should be OK.

> Looking back at that, maybe it just meant that you wouldn't /create_table and /latest or each 10000 row chunk. Should I expect any other behavior? And are there any more improvement expected?

We can't remove the create_table because one may add/remove sensors while the study is ongoing. If we did this only once when joining a study, we would not get the table created. We now only use the create_table and latest check *only once per sync*. One future improvement is to trigger the create table in these cases using MQTT, for example, or only if the study config has changed.

denzilferreira, 2016-09-04:
Hum... this clearing of the old data is more complicated than what it appears to be.

There are sensors and plugins that rely on the existence of records to operate. For example, we capture the sensor profile once, when the sensor is started and there is no profile data yet. Similarly, the Google Auth plugin asks the user to sign in if we don't have their profile info yet. If we clear these tables, we get new rows all the time and requests to authenticate. It won't be manageable to add tables to an exception table either...

commit 4b8a8f3d (2016-09-04): Fixed: #43
- index condition which would not clear old records from the device.
- missing sync of sensor_x data from the device when data was collected before joining study timestamp.
closed 2016-09-04 by denzilferreira