denzilferreira/aware-client #46: Buffering sensor events: API level, timestamps

rkdarst, 2016-09-18:
See #30 for discussion.

Topics to be addressed:
- Currently using a function added in API level 19, this needs to be made backwards-compatible
- Buffering currently can be made more flexible: right now reporting latency may not be set correctly.
- If it was set correctly, we would need to resolve the meaning of sensor timestamps, since it is not standardized across devices.

rkdarst, 2016-12-20:
I am looking at this again.  (could you reopen issue, I can't).  I am using accelerometer as an example.  There is an easy and hard solution.

The easy solution requires adding another database column and handling things on the server side.  For this solution, I add a "sensor_timestamp" column and that column records the sensor timestamp as given by the buffering API.  This doesn't always mean real unix time, it could be time since boot-up or other interpretations.  It is up to the server to correct it (in my case, this is done upon data extraction).  Should there be any problem with a new database column, that defaults to all nulls?

The slighly more complex solution is to emit one test event, and then use that to infer the timestamp units.

Thanks,

- Richard

rkdarst, 2016-12-21:
Hi,

So I have implemented the "easy" strategy here, for accelerometer only so far:

https://github.com/CxAalto/aware-client/commit/8221a085378d2c0c064cle7c522f51cfa949733562d
https://github.com/CxAalto/aware-client/commits/sensor-buffer-io

It adds a column sensor_timestamp, and that contains the raw timestamp values from the sensor event.  The server would have to be clever and turn them into real timestamps, probably assuming the last event in a batch is close to real time.  This strategy doesn't quite match the spirit of Aware being mostly client-side, but it works for now.  (don't pull this yet, needs more testing and improvement)

It would help to get the phone boot time if we wanted to be less heurestic.  This could either be on one one of the aware metadata tables (that should be sent at least once per boot), or could be on the sensor_accelerometer table (is this sent once per boot?).

denzilferreira, 2019-10-01:
Will be addressed on kotlin library.

commit 84634f7b (2016-10-18): - New: log sensors and plugins statuses.
- Fix: #70 - multiple overlay intros.
- Fix: #46 - sensor buffer and FIFO timestamp revert.
closed 2019-10-01 by denzilferreira