denzilferreira/aware-client #47: Periodically updating configuration

rkdarst, 2016-09-18:
Hi,

I am interested in making a way to have configuration periodically updated (at a certain frequency, without MQTT).

Currently in `aware-core / utils.StudyUtils` `onHandleIntent`, I have made a branch which only updates the config.  It is triggered by if an extra attribute is true`is_dynamic_update = intent.getBooleanExtra(EXTRA_DYNAMIC_UPDATE, false);`.  I am trying to see how to schedule this using Scheduler, which is a bit trick to me so far.  Do you have any ideas?

I get can work myself down to here, in `Aware.onStartCommand`:
```
Scheduler.Schedule schedule = new Scheduler.Schedule(SCHEDULE_CONFIG_UPDATE)
    .setActionType(Scheduler.ACTION_TYPE_BROADCAST)
    //.setActionClass(JoinStudy)
    .addActionExtra(StudyUtils.EXTRA_DYNAMIC_UPDATE, true)
    .setInterval(frequency_update);
```

Any suggestions on where to go from there?

There may be an entirely better way to do this, if you would like to provide another idea or would like to hack something together, please let me know.  I have a config option `frequency_webservice_update`.

By the way.. I see `utils/StudyUtils`, which has methods of joining a study.  But I also see `Aware.Aware.JoinStudy`, extending the other.  Which should be used for updating config?

Thanks,
- Richard

denzilferreira, 2016-09-20:
Hi Richard,

The difference between those two is what happens at the end. The StudyUtils is used internally by the client to apply the settings and refresh the UI, while the Aware.Aware.JoinStudy is what is executed when you use Aware.joinStudy() function from a library or plugin for example. 

Perhaps instead of setting a scheduler as a broadcast, you could use create a new inner service in the Aware class, e.g., CheckStudy. Then you don't need that extra attribute at all. You can schedule it as such:

`
Scheduler.Schedule check = Scheduler.getSchedule(SCHEDULE_CONFIG_UPDATE);

if (check == null) { //otherwise you'll schedule this over and over again.

check = new Scheduler.Schedule(SCHEDULE_CONFIG_UPDATE)
.setActionType(Scheduler.ACTION_TYPE_SERVICE)
.setActionClass('com.aware.phone/com.aware.phone.Aware.CheckStudy')
.setInterval(frequency_update);

Scheduler.saveSchedule(check);
}
`

denzilferreira, 2016-10-06:
Working on this.

denzilferreira, 2016-10-15:
Update on this:
We now keep a record of the study info in the aware_studies table in Aware's main content provider. All is missing is checking the config from the server and then detect if there are differences between the two and update accordingly on the phone. Right now, the function we use to check the config, reassigns the user with new MQTT credentials, which is not ideal. I'm adding a new function on the server side to get the current config, without necessarily getting a new MQTT credential combo.

rkdarst, 2016-12-13:
Hi,

I was wondering about this again?  If I remember right, this could be done with the scheduler.  If so, what is the intent to call, com.aware.utils.StudyUtils ?  I can try it and report the config needed.

Thanks,

- Richard

rkdarst, 2016-12-13:
So this config seems to work to make the config update (Note: Python syntax, not JSON, and my server processes this into the form required for AWARE, but you should get the idea):

```
    update_config = {
        "package": "com.aware.phone",
        "schedule": {
            "action": {
                "class": "com.aware.phone/com.aware.utils.StudyUtils",
                "extras": {
                    "study_url": device.qrcode_url(),
                    },
                "type": "service",
                },
            "schedule_id": "update_config",
            "trigger": {
                "interval": 1
                },
            }
        }
```

There is one problem: it seems the schedule is immediately triggered after the config is set.  This causes it to immediately get the config again... causing another update to immediately happen, and so on forever.  I guess last_triggered is reset each time schedule is rebuilt.

One solution would be to set it to update only on certain hours, but: would it still repeat during those hours, and all clients would update at around the same time.  What do you think is the best solution here?

Thanks,

- Richard

rkdarst, 2016-12-19:
Hi,

CxAalto/aware-client@579d20daf19c5907fc317981c1d210b5bceca906

By the way, I added this commit which minimally solves the problem.  I wouldn't use it yet, since I think there may be some new problems, and also there is probably a better way to do it (like updating without triggering schedules immediately).

denzilferreira, 2016-12-20:
Hi Richard,

Yes, this is work in progress which I haven't been able to complete. We do keep track now of the study status. If closed, the client is reset. Missing the parameters from the server. However, recently we made the client open-ended (the user can always change what is shared) by default. This means, the config on the server side can be different from the one set on the client. Which makes this whole config sync pointless.

rkdarst, 2016-12-20:
Hi,

The config updating I implemented at least seems to be working so far.  This, combined with server-side random timestamps, at least minimally fits our needs.

As for "open by default", looking at the commit it seems like it removes any ability to lock the studies.  Will locking be supported in the future?  In the future, what is the plan for this type of thing?  Creating one's own dedicated app?

Thanks,

- Richard

denzilferreira, 2016-12-21:
Yes, a standalone application with AWARE as a library to lock the settings.

denzilferreira, 2017-02-16:
Fixed in development branch in https://github.com/denzilferreira/aware-client/commit/e219279a23203eb127a3466ddce5d70538004023

closed 2017-02-16 by denzilferreira