denzilferreira/aware-client #52: Permissions being asked multiple times

rkdarst, 2016-09-20:
Hi,

Sometimes, when Aware asks for permissions, it gets into a loop and does so multiple times.  Recently, I was demonstrating it to someone and it asked for call/sms/and a few other permissions 5-10 times in a loop.

I don't have details right now, but I am making the issue for tracking, and will update as I learn more.

vedantswain, 2016-09-20:
I faced the same problem just a moment back. About 3-4 permissions are asked in a loop. However this didn't happen when I tried it on Android 5.1, I've only come across this while trying to use it on Android 6

denzilferreira, 2016-09-20:
Yes the issue is only visible starting Marshmallow. I'll look into fixing this as soon as possible.

rkdarst, 2016-09-20:
@denzilferreira , by the way, if you are investigating these type of things, when the app is first installed, sometimes it gives the "click here to join a study" "explore settings(?)" overlays twice.  as in A, B, A, B with the background getting lighter partway through, so that it looks like there are two overlays overlaid.

I've seen it many times when reinstalling in the last few weeks, but I think it may not be every time.  I haven't investigated just yet, so I haven't made a separate issue.

Thanks,
- Richard

rkdarst, 2016-09-23:
Re: the two help overlays being shown multiple times.  I see they are called from `Aware_Client`, `defaultSettings`.  This gets called from both `onStart` and `onPreferenceTreeClick`.  Could this be the cause?  Could those two `Views` be moved to `onStart`?  (I don't know enough to know what the situation is, but when I get time I will try it out.)

rkdarst, 2016-09-23:
Re: preferences, I notice the aware_preferences.xml in core and phone are quite similar but seem to have gotten out of sync.  I see that the aware-phone one has the icons defined, which I guess should not be in core.

* phone has toolbar headings - can this be added to core?
* some of the options are the same but reordered - I will fix these.

Besides the icons, what *should* be different between the two?

denzilferreira, 2016-09-23:
Hi Richard,

On the core, we don’t need the toolbar headings or the icons. That makes the whole library smaller by not having the need to have the drawables bundled. All the rest should be in sync.

Thanks!
Denzil;

denzilferreira, 2016-09-23:
Hum… the defaultSettings makes sure the UI is consistent when we change a setting (by toggling the icons’, etc). Try moving the dialog request to onResume().

Thanks!
Denzil;

rkdarst, 2016-09-23:
RFC re: settings ordering

Right now "developer options" are the first thing in the settings, and communication is way at the bottom.  As part of this change, I can do some other reordering.  I think we should assume the average person is a user, not developer.

So: what options are most important and should be at the top?

I am thinking now: At top, deviceid, device label, AWARE version, then Communications section, then "Advanced options", then all the sensors.

* What else is high priority or should go at the top?
* Can "lightweight i/o" be moved to advanced settings?
* RFC: AWARE Study -> Server connection (?)
* I'll reorder sensors to alphabetical order, there are a few quirks there.
* "Mobile ESM/EMA" is currently sorted under "ESM".  Could acronyms be expanded and/or renamed Questionnaires with Mobile ESM/EMA as subheading?

I'm asking because I only know my own use case, not what everyone does.

denzilferreira, 2016-10-11:
@rkdarst this multiple overlay (study info, permissions request) is no longer an issue as we redesigned joining a study from the client with a wizard, set by set setup automated flow.

Regarding the order of the options, the current changes are ok and merged, so closing this for now.

closed 2016-10-11 by denzilferreira