denzilferreira/aware-client #53: URI schema for linking

rkdarst, 2016-09-22:
Hi,

I have a new user story.  In this case, subjects will come and attend a meeting introducing the study.  Then, they will link devices there.  Some may just have the phone, and not a computer too.  They will have  their own accounts on the study server, which will display the instructions.  If they log in to the study site on their phone, they can't scan the QR code with their own phone.  I'd like to be able to make a button with a URI that gets handled by the app and does the linking.

- Does this already exist?  (I couldn't find any intent filters in the manifests)
- What should the scheme be?  `aware:` seems too generic, `awareregister:` or `awarefw-register` ?
- @denzilferreira, any suggestions about what class should handle the uri?  should it go to one of the StudyJoin or StudyUtils classes as mentioned in another issue?
- @tetujin, any comments from the iOS side?

Thanks,
- Richard

denzilferreira, 2016-09-22:
Maybe a different approach? Not sure how to handle custom URI schemes on Android.

I’ve been thinking about having another alternative to the QRCode, in which the participant simply inputs the study API key. This would load from the server the description of the study and they can sign-up directly from the client UI. They would not necessarily need to have an account on the study server a priori.

Denzil;

rkdarst, 2016-09-22:
A study key would be interesting, but it would need to be able to direct to the right server somehow, if there is more than one.  So that would end up including a domain name...  Right now the "manual" way to join is copying and pasting the webservice URL. At least we could add a action at the top to "Enter study key/url" and then handle it.  It can start out with the study URL, and could be adapted to other types of codes.

Custom URI schemes is not that hard though:
[SO#1](http://stackoverflow.com/questions/2448213/how-to-implement-my-very-own-uri-scheme-on-android), [SO#2, including iOS](http://stackoverflow.com/questions/11421048/android-ios-custom-uri-protocol-handling#11421251)

For any Activity, we add an intent filter:
```
<intent-filter>
  <action android:name="android.intent.action.VIEW" /> 
  <category android:name="android.intent.category.DEFAULT" /> 
  <category android:name="android.intent.category.BROWSABLE" /> 
  <data android:scheme="aware-register" /> 
</intent-filter>
```
And then if such a URI is clicked, it starts the activity with the URI in the intent, which can be parsed and handled.

denzilferreira, 2017-07-11:
Fixed in latest, done just right. It supports two kinds of schemas:
aware://path_to_study
aware-ssl://path_to_study

Where path_to_study is without http:// or https://.

commit 9d172e34 (2017-07-11): - Add support to aware:// and aware-ssl:// study URLs. This allows enrolling to a study on devices without a camera or QRCode!
- Fixed #53 @rkdarst
closed 2017-07-11 by denzilferreira