denzilferreira/aware-client #69: How to have unlocked studies?

rkdarst, 2016-10-13:
Hi,

In our use case, we sometimes allow users to have their device linked (to send data to the webservice), but we don't specify configuration for them - we let them turn on and off sensors as they would like.  This was nicely handled before, since `Aware.isStudy()` is what locked everything, and if we could set a null `study_id`, and then users could freely test things.

In the latest commits (7cba434, cd1059b), this is no longer the case and we have this code instead:

```
    public static boolean isStudy(Context c) {
        boolean participant = false;
        Cursor study = c.getContentResolver().query(Aware_Provider.Aware_Studies.CONTENT_URI, null,
                Aware_Provider.Aware_Studies.STUDY_URL + " LIKE '" + Aware.getSetting(c, Aware_Preferences.WEBSERVICE_SERVER) +
                        "' AND " + Aware_Provider.Aware_Studies.STUDY_JOINED + ">0" +
                        " AND " + Aware_Provider.Aware_Studies.STUDY_EXIT + "=0",
                null, null);
        if (study != null && study.moveToFirst())
            participant = true;
        if (study != null && !study.isClosed()) study.close();
        return participant;
    }
```

What od you think of this?
- What would you recommend to allow unlocked studies?
- Should the definition of `isStudy()` be "in study that handles all configuration", as opposed to "currently uploading data"?

This is same as in CxAalto/aware-client#6 and was fixed there.

Thanks,
- Richard

denzilferreira, 2016-10-15:
Hum... i didn't think of this scenario. I agree, this is something nice to have, i.e., allowing participants to pick whatever they wish to share in terms of data. I can take a look at this during your visit.

rkdarst, 2016-10-16:
The way it worked before did this quite well.  Looking at the code, almost all usages of isStudy are for UI-related things, except for WebserviceHelper(questionable), Mqtt, Aware_Activity.  We could just add either isLocked/isChangeAllowed or as isLinked, and change the relevant ones.  The study_id test seemed to work before, if that is still used.

Also, now even though I am linked to the webservice, I can change all values in the UI.  I haven't studied in depth, but perhaps there is something going wrong here.

rkdarst, 2016-12-13:
Is working for now...

closed 2016-12-13 by rkdarst