denzilferreira/aware-client #72: Aware crashes

rkdarst, 2016-10-14:
Hi,

For the last day (ever since I updated, previous update was a week or so ago so this could be in any of the code of the last week), I have gotten very frequent crashes.  There may be more/other things, but here is the main crash (I think):

```
10-15 01:06:49.553 14328-14328/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                   Process: com.aware.phone, PID: 14328
                                                   java.lang.RuntimeException: Error receiving broadcast Intent { act=ACTION_AWARE_SYNC_DATA flg=0x10 } in com.aware.Aware$Aware_Broadcaster@40ce47c
                                                       at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:891)
                                                       at android.os.Handler.handleCallback(Handler.java:739)
                                                       at android.os.Handler.dispatchMessage(Handler.java:95)
                                                       at android.os.Looper.loop(Looper.java:148)
                                                       at android.app.ActivityThread.main(ActivityThread.java:5417)
                                                       at java.lang.reflect.Method.invoke(Native Method)
                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
                                                    Caused by: java.lang.ArrayIndexOutOfBoundsException: length=2; index=2
                                                       at com.aware.Aware$Aware_Broadcaster.onReceive(Aware.java:1529)
                                                       at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:881)
                                                       at android.os.Handler.handleCallback(Handler.java:739) 
                                                       at android.os.Handler.dispatchMessage(Handler.java:95) 
                                                       at android.os.Looper.loop(Looper.java:148) 
                                                       at android.app.ActivityThread.main(ActivityThread.java:5417) 
                                                       at java.lang.reflect.Method.invoke(Native Method) 
                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726) 
                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616) 
```

Relevant line:
```
                    webserviceHelper.putExtra(WebserviceHelper.EXTRA_CONTENT_URI, CONTEXT_URIS[i].toString());
```

Context in Aware.py:
```
            if (intent.getAction().equals(Aware.ACTION_AWARE_SYNC_DATA) && Aware.getSetting(context, Aware_Preferences.STATUS_WEBSERVICE).equals("true")) {
                for( int i=0; i<DATABASE_TABLES.length; i++ ) {
                    Intent webserviceHelper = new Intent(context, WebserviceHelper.class);
                    webserviceHelper.setAction(WebserviceHelper.ACTION_AWARE_WEBSERVICE_SYNC_TABLE);
                    webserviceHelper.putExtra(WebserviceHelper.EXTRA_TABLE, DATABASE_TABLES[i]);
                    webserviceHelper.putExtra(WebserviceHelper.EXTRA_FIELDS, TABLES_FIELDS[i]);
                    webserviceHelper.putExtra(WebserviceHelper.EXTRA_CONTENT_URI, CONTEXT_URIS[i].toString());
                    context.startService(webserviceHelper);
                }
            }
```

I am submitting this quickly now, I will check more when I can, but someone else may be able to know the problem sooner.  I'm not exactly the sure the point of CONTEXT_URIS.  There may also be other errors, in which case I will update this issue.

Thanks,
- Richard

denzilferreira, 2016-10-15:
Update your sources to the latest. This was fixed two days ago or so :)

rkdarst, 2016-10-16:
Yes, seems to work now.  I thought I was on play store version, and I switched to self-compiled but something was still weird.  No crashes today, I hope it was just me.  By the way: could we add the git hash into the app description?  Would help to avoid this.  I see at least http://stackoverflow.com/a/25707575, not sure the best way.

closed 2016-10-16 by rkdarst