denzilferreira/aware-client #88: Certificate being downloaded all the time

rkdarst, 2016-12-13:
Hi,

Since approximately f39b3d8 (Ion cache disabled), the client re-downloads the certificate on every request.  This was the point of this commit, but seems to defeat the purpose of pinning a certificate.  The goal should be to ensure data confidentiality by authorizing the server.  If it is download insecurely with no verification every time data is uploaded, then security is basically like using a self-signed cert with no verification at all.  (in both cases, it would take someone manipulating traffic to intercept it, with the difference in difficulty rather small).  This is quite a problem for us, since we need to say our app is secure for our projects.

The previous strategy, of downloading only once insecurely, at least minimized the vulnerable period so made it OK.  But I understand the problem with certificate migration that lead to this.  So the question is what to do now.

- My exciting QR-code based certificates at least should have disabled this behavior, but it seems that the download is being forced (I am not sure where).  It doesn't provide a key migration strategy, but it would provide security during a single project.
- There could be different generations of endpoints: 1.api.project.com, 2.api.project.com, etc.  Each uses a new certificate, as old projects go out.  Not very satisfying.
- Some key management strategy where new keys can be sent via config file (requires periodic config updating).  Also requires multi-keying and complexity inside the code, and ability to update in advance.
- Some of these could be combined.

I don't see a very good option, but what do you think?

As a work-around, I have tried commenting out the downloadCertificate lines in getHttps and getCertificate, however it is still fetching the certificate every time.  Do you know why it is doing this?  By the way, on my last edit, downloadCertificate would always unconditionally download the new crt, but handleUrl would download if needed, but if URL parameters were present which meant that only those were used and didn't do the insecure downloading.  If this worked, it would provide the same default, but also an option for more secure uploading that could be enabled.

Thanks,

- Richard

rkdarst, 2016-12-13:
Update: now it is using the QR code embedded within the URL only, so it does the right thing.  I am sure I had updated the app before, so I am not sure what happened differently.  At least the base secure case works again.

Also, if the config is periodically updated (#47), then in theory certificates can be rotated by having the config on (n-1).api.site.com point to API n.api.site.com with the necessary certificate information.  This could become a bit fragile if errors aren't handled well.

Thanks,

- Richard

denzilferreira, 2016-12-20:
Hi Richard,

I need to spend some time in this. I don't like the way the client is performing now (always downloading the certificate) either.

What I've been thinking of is:
- Certificate is always downloaded only if you join a study for the first time
- Certificate is cached and we keep track of the expiring date. Once the expired date arrives, the client will attempt to download new certificate from the same place (it should be then a newer certificate)
- Certificate is updated and data collection resumes

commit a7fccf5b (2017-02-27): - Fixed #88
- Certificate is shared among plugins. Downloaded when joining study, reutilised after for every HTTPS request.
closed 2017-02-27 by denzilferreira