denzilferreira/aware-client #104: Bluetooth is not recording data every 60 sec

JulioV, 2017-02-08:
When on a study with the following sensors activated Accelerometer, Magnetometer: Applications, Processor, Light, Timezone, Battery, Communication, ESM, MQTT, Locations, Network, Bluetooth, and WiFi. The Bluetooth sensors does not record 60 times per hour but WiFi sensor does.

Bluetooth (second column is records per hour)
29c9acb3-7a19-4a29-adfe-319a03efc064    7th 18h    11    41
29c9acb3-7a19-4a29-adfe-319a03efc064    7th 19h    40    65
29c9acb3-7a19-4a29-adfe-319a03efc064    7th 20h    19    26
29c9acb3-7a19-4a29-adfe-319a03efc064    7th 21h    21    27
29c9acb3-7a19-4a29-adfe-319a03efc064    7th 22h    22    25
29c9acb3-7a19-4a29-adfe-319a03efc064    7th 23h    11    14
29c9acb3-7a19-4a29-adfe-319a03efc064    8th 00h    18    27
29c9acb3-7a19-4a29-adfe-319a03efc064    8th 01h    21    29
29c9acb3-7a19-4a29-adfe-319a03efc064    8th 02h    16    18
29c9acb3-7a19-4a29-adfe-319a03efc064    8th 03h    16    18
29c9acb3-7a19-4a29-adfe-319a03efc064    8th 04h    11    17

Wifi (second column is records per hour)
'29c9acb3-7a19-4a29-adfe-319a03efc064', '7th 18h', '15', '950'
'29c9acb3-7a19-4a29-adfe-319a03efc064', '7th 19h', '60', '3891'
'29c9acb3-7a19-4a29-adfe-319a03efc064', '7th 20h', '60', '4634'
'29c9acb3-7a19-4a29-adfe-319a03efc064', '7th 21h', '60', '4677'
'29c9acb3-7a19-4a29-adfe-319a03efc064', '7th 22h', '60', '4626'
'29c9acb3-7a19-4a29-adfe-319a03efc064', '7th 23h', '60', '4755'
'29c9acb3-7a19-4a29-adfe-319a03efc064', '8th 00h', '60', '4723'
'29c9acb3-7a19-4a29-adfe-319a03efc064', '8th 01h', '60', '4371'
'29c9acb3-7a19-4a29-adfe-319a03efc064', '8th 02h', '53', '4289'
'29c9acb3-7a19-4a29-adfe-319a03efc064', '8th 03h', '60', '4829'
'29c9acb3-7a19-4a29-adfe-319a03efc064', '8th 04h', '23', '1592'

JulioV, 2017-02-09:
I did more testing. I have to phones of the same model, one with the latest dev core and one with the same version but using a modified version of the Bluetooth class using the Scheduler instead of the alarmManager.

These are the recordings from the latest dev core with alarmManager without plugins:

# device_id, hour, minutesWithRecords, recordsPerHour
96357276-f2b5-4997-97fe-8d7f7c4387c0, 8th 20h, 60, 115
96357276-f2b5-4997-97fe-8d7f7c4387c0, 8th 21h, 57, 109
96357276-f2b5-4997-97fe-8d7f7c4387c0, 8th 22h, 58, 122
96357276-f2b5-4997-97fe-8d7f7c4387c0, 8th 23h, 55, 108
96357276-f2b5-4997-97fe-8d7f7c4387c0, 9th 00h, 59, 112
96357276-f2b5-4997-97fe-8d7f7c4387c0, 9th 01h, 54, 108
96357276-f2b5-4997-97fe-8d7f7c4387c0, 9th 02h, 59, 104
96357276-f2b5-4997-97fe-8d7f7c4387c0, 9th 03h, 55, 110
96357276-f2b5-4997-97fe-8d7f7c4387c0, 9th 04h, 57, 101

These are the results of the latest core with the Scheduler with plugins:

# device_id, hour, minutesWithRecords, recordsPerHour
29c9acb3-7a19-4a29-adfe-319a03efc064, 8th 20h, 44, 77
29c9acb3-7a19-4a29-adfe-319a03efc064, 8th 21h, 51, 90
29c9acb3-7a19-4a29-adfe-319a03efc064, 8th 22h, 53, 98
29c9acb3-7a19-4a29-adfe-319a03efc064, 8th 23h, 49, 80
29c9acb3-7a19-4a29-adfe-319a03efc064, 9th 00h, 53, 89
29c9acb3-7a19-4a29-adfe-319a03efc064, 9th 01h, 55, 96
29c9acb3-7a19-4a29-adfe-319a03efc064, 9th 02h, 51, 94
29c9acb3-7a19-4a29-adfe-319a03efc064, 9th 03h, 47, 80

And for comparison purposes these are the results of the latest core with alamManager with plugins:
29c9acb3-7a19-4a29-adfe-319a03efc064 7th 18h 11 41
29c9acb3-7a19-4a29-adfe-319a03efc064 7th 19h 40 65
29c9acb3-7a19-4a29-adfe-319a03efc064 7th 20h 19 26
29c9acb3-7a19-4a29-adfe-319a03efc064 7th 21h 21 27
29c9acb3-7a19-4a29-adfe-319a03efc064 7th 22h 22 25
29c9acb3-7a19-4a29-adfe-319a03efc064 7th 23h 11 14
29c9acb3-7a19-4a29-adfe-319a03efc064 8th 00h 18 27
29c9acb3-7a19-4a29-adfe-319a03efc064 8th 01h 21 29
29c9acb3-7a19-4a29-adfe-319a03efc064 8th 02h 16 18
29c9acb3-7a19-4a29-adfe-319a03efc064 8th 03h 16 18
29c9acb3-7a19-4a29-adfe-319a03efc064 8th 04h 11 17

It seems that the plugins interfere somehow with the alarmManager making the BT sensing intermittent, this might not be the case for WiFI due to the recent changes where we made WiFi sensing async. The Scheduler manager seems to be a better solution but not perfect.

JulioV, 2017-02-13:
I tried sensing BT data with the Scheduler for 3 days in a row and the problem persists, we are only getting around 20 sensings per hour.

denzilferreira, 2017-02-16:
Fixed in #115

closed 2017-02-16 by denzilferreira