denzilferreira/aware-client #107: Client crashes when tapping in Stream

JulioV, 2017-02-08:
The client crashes after clicking in Stream if there are any active plugins:

`2-06 15:43:00.832 8392-8392/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                 Process: com.aware.phone, PID: 8392
                                                 java.lang.NullPointerException: Attempt to invoke virtual method 'int android.view.View.getImportantForAccessibility()' on a null object reference
                                                     at android.widget.AbsListView.obtainView(AbsListView.java:2363)
                                                     at android.widget.ListView.makeAndAddView(ListView.java:1864)
                                                     at android.widget.ListView.fillDown(ListView.java:698)
                                                     at android.widget.ListView.fillFromTop(ListView.java:759)
                                                     at android.widget.ListView.layoutChildren(ListView.java:1673)
                                                     at android.widget.AbsListView.onLayout(AbsListView.java:2151)
                                                     at android.view.View.layout(View.java:15689)
                                                     at android.view.ViewGroup.layout(ViewGroup.java:5040)
                                                     at android.widget.RelativeLayout.onLayout(RelativeLayout.java:1076)
                                                     at android.view.View.layout(View.java:15689)
                                                     at android.view.ViewGroup.layout(ViewGroup.java:5040)
                                                     at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1703)
                                                     at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1557)
                                                     at android.widget.LinearLayout.onLayout(LinearLayout.java:1466)
                                                     at android.view.View.layout(View.java:15689)
                                                     at android.view.ViewGroup.layout(ViewGroup.java:5040)
                                                     at android.widget.FrameLayout.layoutChildren(FrameLayout.java:579)
                                                     at android.widget.FrameLayout.onLayout(FrameLayout.java:514)
                                                     at android.view.View.layout(View.java:15689)
                                                     at android.view.ViewGroup.layout(ViewGroup.java:5040)
                                                     at android.support.v7.widget.ActionBarOverlayLayout.onLayout(ActionBarOverlayLayout.java:437)
                                                     at android.view.View.layout(View.java:15689)
                                                     at android.view.ViewGroup.layout(ViewGroup.java:5040)
                                                     at android.widget.FrameLayout.layoutChildren(FrameLayout.java:579)
                                                     at android.widget.FrameLayout.onLayout(FrameLayout.java:514)
                                                     at android.view.View.layout(View.java:15689)
                                                     at android.view.ViewGroup.layout(ViewGroup.java:5040)
                                                     at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1703)
                                                     at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1557)
                                                     at android.widget.LinearLayout.onLayout(LinearLayout.java:1466)
                                                     at android.view.View.layout(View.java:15689)
                                                     at android.view.ViewGroup.layout(ViewGroup.java:5040)
                                                     at android.widget.FrameLayout.layoutChildren(FrameLayout.java:579)
                                                     at android.widget.FrameLayout.onLayout(FrameLayout.java:514)
                                                     at android.view.View.layout(View.java:15689)
                                                     at android.view.ViewGroup.layout(ViewGroup.java:5040)
                                                     at android.view.ViewRootImpl.performLayout(ViewRootImpl.java:2116)
                                                     at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:1873)
                                                     at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1084)
                                                     at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:5990)
                                                     at android.view.Choreographer$CallbackRecord.run(Choreographer.java:767)
                                                     at android.view.Choreographer.doCallbacks(Choreographer.java:580)
                                                     at android.view.Choreographer.doFrame(Choreographer.java:550)
                                                     at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:753)
                                                     at android.os.Handler.handleCallback(Handler.java:746)
                                                     at android.os.Handler.dispatchMessage(Handler.java:95)
                                                     at android.os.Looper.loop(Looper.java:135)
                                                     at android.app.ActivityThread.main(ActivityThread.java:5343)
                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                     at java.lang.reflect.Method.invoke(Method.java:372)
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:907)
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:702)`

denzilferreira, 2017-02-16:
I'm trying to replicate this. I get no crash when pressing the Stream and I have active plugins. Could this be a faulty plugin (e.g., some xml problem on the layout)? The Stream loads the ContextCard class. If your plugin does not make use of the ContextCard, you can safely delete it from the project.

denzilferreira, 2017-02-16:
Nevermind, managed to replicate. Fixing.

commit 2e0dd58e (2017-02-16): Fix #107
closed 2017-02-16 by denzilferreira