denzilferreira/aware-client #111: Schedulers in the same hour/month/year fail

JulioV, 2017-02-08:
When setting up multiple schedulers on the same hour/month/year only the first one will be executed due to this condition being true:

https://github.com/denzilferreira/aware-client/blob/development/aware-core/src/main/java/com/aware/utils/Scheduler.java#L1189

closed 2017-02-14 by denzilferreira

rkdarst, 2017-05-17:
For this issue...

Now, if you have a scheduler for `hour=[22]`, for example, it will run *every* time the schedule runner activates during that hour, instead of just once.  Is this the expected behavior?  In my case, this caused ESMs scheduled during that hour to repeat over and over.

Since I got them now anyway, here is is before the change: https://github.com/denzilferreira/aware-client/blob/b37244a%5E/aware-core/src/main/java/com/aware/utils/Scheduler.java#L1189

And after change: https://github.com/denzilferreira/aware-client/blob/b37244a/aware-core/src/main/java/com/aware/utils/Scheduler.java#L1195

I see the problem with the original case, though: hour=[22] and minute=[0,15,30,45] would cause it to run only once during hour=22.  I guess some solution like "use the` is_same_*` condition only on the most fine-grained of all of the conditions?

Abstractly I can work on this, but not before next week.  You could reopen and assign to me if you wanted.

NBerkel, 2017-05-18:
I cannot replicate this issue on my phone - I have a random scheduler configured according to the following code, and it fires only once during the configured hour. Are you sure that the schedule itself changes? AWARE prints to the logcat from time to time, but the original scheduler configuration remains intact.

--

            Scheduler.Schedule random = new Scheduler.Schedule("testRandom");
            random.addHour(17)
                    .random(1, 15) //5 randoms, at least 15 minutes apart
                    .setActionType(Scheduler.ACTION_TYPE_SERVICE)
                    .setActionClass(c.getPackageName() + "/" + Aware_TTS.class.getName())
                    .addActionExtra(Aware_TTS.EXTRA_TTS_TEXT, "Random triggered!")
                    .addActionExtra(Aware_TTS.EXTRA_TTS_REQUESTER, c.getPackageName());

            Scheduler.saveSchedule(c, random);

rkdarst, 2017-05-18:
Randoms use a timer scheduler behind the scenes. So I don't think it would
manifest in this case.

Try scheduling for only hour=anything, using latest development in this
repo.

NBerkel, 2017-05-18:
Thank you for your reply. I tested with the following code, which triggered only once (at the start of the hour) - so unfortunately I was not able to replicate using this approach either.

 Scheduler.Schedule timer = new Scheduler.Schedule("test_scheduler");
            timer.addHour(18)
                    .setActionType(Scheduler.ACTION_TYPE_SERVICE)
                    .setActionClass(c.getPackageName() + "/" + Aware_TTS.class.getName())
                    .addActionExtra(Aware_TTS.EXTRA_TTS_TEXT, "Random triggered!")
                    .addActionExtra(Aware_TTS.EXTRA_TTS_REQUESTER, c.getPackageName());

rkdarst, 2017-05-18:
Hm....

And this is using the denzilferreira/aware-client, not CxAalto/aware-client repository?  In the CxAalto one, I have a workaround (reverting the patch) that I'll upgrade to the better fix (only use most granular one).

If you are using denzilferreira/aware-client, do you have any idea what limits it to running only once in the hour?

Thanks,

- Richard

denzilferreira, 2017-05-18:
We do check the last timestamp. I think it's because you are cleaning the database when syncing.

Sent from VMware Boxer

rkdarst, 2017-05-18:
Is this in ESMs or scheduler?  In the scheduler, I see that the last timestamp is only used for the timer and interval schedulers.  It used to be used for all the others, but that was removed in this patch.  If there is also an interval spaced, they will be combined so that it's not executed too often.  At least this is my analysis so far...

I'll try checking more tonight.

Our code never deletes from the scheduler DB.  However, it does delete from the ESMs DB, and this was the original problem with our app (HYKS): previously the scheduler ran several times an hour, but the middle layer would look and see if a question has been already answered before making a new ESM, thus preventing the duplicates.

rkdarst, 2017-05-18:
Though, I really like the idea of the asking the ESMs multiple times until they are answered...

commit b37244a3 (2017-02-14): Fixed #110: schedulers with multiple copies of the same element
Fixed #111: schedulers in the same hour would fail
Fixed #114: unclosed cursor in Scheduler