denzilferreira/aware-client #113: WiFi service is crashing

JulioV, 2017-02-08:
The exception is caught but there is a problem on how WiFi logging is managed.

`2-08 20:01:04.234 8701-8716/? W/SQLiteCompiledSql: Releasing statement in a finalizer. Please ensure that you explicitly call close() on your cursor: INSERT OR IGNORE  INTO wifi(ssid, device_id, bssid, rssi, security, timestamp, frequency) VALUES(?, 
                                                    net.sqlcipher.database.DatabaseObjectNotClosedException: Application did not close the cursor or database object that was opened here
                                                        at net.sqlcipher.database.SQLiteCompiledSql.<init>(SQLiteCompiledSql.java:62)
                                                        at net.sqlcipher.database.SQLiteProgram.<init>(SQLiteProgram.java:83)
                                                        at net.sqlcipher.database.SQLiteStatement.<init>(SQLiteStatement.java:39)
                                                        at net.sqlcipher.database.SQLiteDatabase.compileStatement(SQLiteDatabase.java:1541)
                                                        at net.sqlcipher.database.SQLiteDatabase.insertWithOnConflict(SQLiteDatabase.java:1965)
                                                        at com.aware.providers.WiFi_Provider.insert(WiFi_Provider.java:215)
                                                        at android.content.ContentProvider$Transport.insert(ContentProvider.java:236)
                                                        at android.content.ContentResolver.insert(ContentResolver.java:1205)
                                                        at com.aware.WiFi$WifiApResults.call(WiFi.java:263)
                                                        at com.aware.WiFi$WifiApResults.call(WiFi.java:238)
                                                        at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                                                        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                                                        at java.lang.Thread.run(Thread.java:818)
02-08 20:01:04.234 8701-8716/? W/SQLiteCompiledSql: Releasing statement in a finalizer. Please ensure that you explicitly call close() on your cursor: INSERT OR IGNORE  INTO sensor_wifi(timestamp, mac_address, ssid, device_id, bssid) VALUES(?, ?, ?, ?
                                                    net.sqlcipher.database.DatabaseObjectNotClosedException: Application did not close the cursor or database object that was opened here
                                                        at net.sqlcipher.database.SQLiteCompiledSql.<init>(SQLiteCompiledSql.java:62)
                                                        at net.sqlcipher.database.SQLiteProgram.<init>(SQLiteProgram.java:83)
                                                        at net.sqlcipher.database.SQLiteStatement.<init>(SQLiteStatement.java:39)
                                                        at net.sqlcipher.database.SQLiteDatabase.compileStatement(SQLiteDatabase.java:1541)
                                                        at net.sqlcipher.database.SQLiteDatabase.insertWithOnConflict(SQLiteDatabase.java:1965)
                                                        at com.aware.providers.WiFi_Provider.insert(WiFi_Provider.java:228)
                                                        at android.content.ContentProvider$Transport.insert(ContentProvider.java:236)
                                                        at android.content.ContentResolver.insert(ContentResolver.java:1205)
                                                        at com.aware.WiFi$WifiInfoFetch.call(WiFi.java:217)
                                                        at com.aware.WiFi$WifiInfoFetch.call(WiFi.java:198)
                                                        at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                                                        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                                                        at java.lang.Thread.run(Thread.java:818)
02-08 20:01:04.237 8701-8716/? W/SQLiteCompiledSql: Releasing statement in a finalizer. Please ensure that you explicitly call close() on your cursor: select count(*) from sqlite_master;
                                                    net.sqlcipher.database.DatabaseObjectNotClosedException: Application did not close the cursor or database object that was opened here
                                                        at net.sqlcipher.database.SQLiteCompiledSql.<init>(SQLiteCompiledSql.java:62)
                                                        at net.sqlcipher.database.SQLiteProgram.<init>(SQLiteProgram.java:83)
                                                        at net.sqlcipher.database.SQLiteQuery.<init>(SQLiteQuery.java:49)
                                                        at net.sqlcipher.database.SQLiteDirectCursorDriver.query(SQLiteDirectCursorDriver.java:42)
                                                        at net.sqlcipher.database.SQLiteDatabase.rawQueryWithFactory(SQLiteDatabase.java:1772)
                                                        at net.sqlcipher.database.SQLiteDatabase.rawQuery(SQLiteDatabase.java:1737)
                                                        at net.sqlcipher.database.SQLiteDatabase.openDatabaseInternal(SQLiteDatabase.java:2354)
                                                        at net.sqlcipher.database.SQLiteDatabase.openDatabase(SQLiteDatabase.java:1101)
                                                        at net.sqlcipher.database.SQLiteDatabase.openDatabase(SQLiteDatabase.java:1068)
                                                        at net.sqlcipher.database.SQLiteDatabase.openDatabase(SQLiteDatabase.java:1017)
                                                        at net.sqlcipher.database.SQLiteDatabase.openDatabase(SQLiteDatabase.java:971)
                                                        at com.aware.utils.DatabaseHelper.getWritableDatabase(DatabaseHelper.java:219)
                                                        at com.aware.providers.WiFi_Provider.initializeDB(WiFi_Provider.java:129)
                                                        at com.aware.providers.WiFi_Provider.insert(WiFi_Provider.java:204)
                                                        at android.content.ContentProvider$Transport.insert(ContentProvider.java:236)
                                                        at android.content.ContentResolver.insert(ContentResolver.java:1205)
                                                        at com.aware.WiFi$WifiInfoFetch.call(WiFi.java:217)
                                                        at com.aware.WiFi$WifiInfoFetch.call(WiFi.java:198)
                                                        at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                                                        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                                                        at java.lang.Thread.run(Thread.java:818)`

denzilferreira, 2017-02-16:
Probably fixed in the latest push ?

JulioV, 2017-03-09:
Probably, I have not seen this in a while

closed 2017-03-10 by denzilferreira