denzilferreira/aware-client #131: Unkown URI error

JulioV, 2017-02-20:
The latest dev was syncing on a study and this exception appeared:

```
02-20 04:42:27.792 5433-5433/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                 Process: com.aware.phone, PID: 5433
                                                 java.lang.RuntimeException: Unable to start service com.aware.Applications@1cb8bd3b with Intent { cmp=com.aware.phone/com.aware.Applications }: java.lang.IllegalArgumentException: Unknown URI content://com.aware.phone.provider.aware/aware_log
                                                     at android.app.ActivityThread.handleServiceArgs(ActivityThread.java:2941)
                                                     at android.app.ActivityThread.access$2100(ActivityThread.java:155)
                                                     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1415)
                                                     at android.os.Handler.dispatchMessage(Handler.java:102)
                                                     at android.os.Looper.loop(Looper.java:135)
                                                     at android.app.ActivityThread.main(ActivityThread.java:5343)
                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                     at java.lang.reflect.Method.invoke(Method.java:372)
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:907)
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:702)
                                                  Caused by: java.lang.IllegalArgumentException: Unknown URI content://com.aware.phone.provider.aware/aware_log
                                                     at com.aware.providers.Aware_Provider.insert(Aware_Provider.java:408)
                                                     at android.content.ContentProvider$Transport.insert(ContentProvider.java:236)
                                                     at android.content.ContentResolver.insert(ContentResolver.java:1205)
                                                     at com.aware.Aware.debug(Aware.java:473)
                                                     at com.aware.Applications.onStartCommand(Applications.java:440)
                                                     at android.app.ActivityThread.handleServiceArgs(ActivityThread.java:2924)
                                                     at android.app.ActivityThread.access$2100(ActivityThread.java:155) 
                                                     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1415) 
                                                     at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                     at android.os.Looper.loop(Looper.java:135) 
                                                     at android.app.ActivityThread.main(ActivityThread.java:5343) 
                                                     at java.lang.reflect.Method.invoke(Native Method) 
                                                     at java.lang.reflect.Method.invoke(Method.java:372) 
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:907) 
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:702) 
```

JulioV, 2017-02-20:
Maybe related to this? [Stackoverflow](http://stackoverflow.com/questions/41474623/sporadic-illegalargumentexception-unknown-url-content)

denzilferreira, 2017-03-09:
Fixed. We shouldn't be starting explicitly the Applications service (it's an accessibility service).

closed 2017-03-09 by denzilferreira