denzilferreira/aware-client #148: Specific StudyUtils code updating (as opposed to replacing) the configuration

rkdarst, 2017-03-05:
Hi,

As you know, I've been working on making a periodically updating config.  Some working scheduler code is in #47 and #147.  I think one issue may be that I use StudyUtils to join a study, which resets Aware before completely re-setting all config.  It seems that we could add a option to make it not call Aware.reset(), and things will probably just work (for example, this could be done as a scheduler extra).  For example, the scheduler is smart enough to know to not create a new schedule with the same ID.

The question here is: is this a good idea?  Does it fit in with the big picture and how Aware is supposed to be used?  Is there a better way?

Pinging @NBerkel also, since we have talked about this.

Thanks,
- Richard

rkdarst, 2017-03-05:
Actually I see a settings update in `AsyncStudyCheck` - and this seems like exactly the right way of doing config updating.  So all of the stuff above doesn't matter.

I see it is hard coded for updating every 5 minutes.  Doesn't this seem a bit too frequent?  I guess it should be a configuration option?

denzilferreira, 2017-03-06:
We check if the study is still ongoing every 5 minutes, where we also get the study configuration setup. 5 minutes seems to be a good compromise between seeing the settings getting applies as soon as possible if the study status changes. Making the waiting period longer will only confuse researchers on why it does not apply the new settings yet...

rkdarst, 2017-03-07:
Hmm... it does keep config updated, but in most of my studies I basically assume things will not be fully in sync.  Phones will be off, not have internet, etc (very long studies in the real world).  While probably not that bad overall, I have been trying to minimize any resource usage possible.

I can hard-code a change in my branch, but I would also like to contribute a configuration option.  The sync schedule is only created once so it needs updating somehow.  In could simply be recreated and saved again, but where?  Or each sync could compare the interval to new interval, and re-save in that case.  Suggestions for what is most Aware-like (or if we should do nothing)?

Thanks,
- Richard

denzilferreira, 2017-03-09:
Richard, if there is no internet, nothing will happen... but you were on to something: this only makes sense if we are in a study, not in the keep alive action. I've moved it to the study compliance check, which occurs every 10 minutes.

denzilferreira, 2017-03-09:
Fixed?

rkdarst, 2017-03-12:
Well, the bigger issue was making too many requests.  Any long-term study we do will have to already take into account that configs may get slightly out of sync.  For now I will handle this by a local change to my branch, but will consider making an upstream change at some time in the future.

closed 2017-03-12 by rkdarst