denzilferreira/aware-client #211: AWARE now always keeps an active alarm?

rkdarst, 2017-07-12:
Hi,

I recently forward-ported my changes to latest aware-client/development, and now I see the OS always has an alarm active for the current minute.  This is indicated, for example, on the lockscreen and in the pull-down system tray where it would normally say "wake-up alarm at 07:00".

I know that some recent changes have been about foregrounding and alarms, so my first thought was this might be related.  I have just seen this and done no other investigation, though.  Does this happen to anyone else?

This should probably be changed if at all possible: this interferes with normal OS (informational) functionality since this seems to always override the real alarm display, and makes it seem that an alarm is always on.

denzilferreira, 2017-07-13:
Yes, this is needed for Android O, to make sure that the scheduler can check for tasks every minute. The only side-effect  is showing an alarm icon to the top bar. For a study/data collection effort, we believe it's an acceptable UI nuisance.

rkdarst, 2017-07-13:
Hi,

So, basically, the foreground persistent notification strategy doesn't
work anymore? (or doesn't work well enough to keep all data coming
in?)

denzilferreira, 2017-07-13:
Yeah, doesn't work reliably... Thanks Google!

Denzil;

rkdarst, 2017-07-13:
Yeah, that's too bad... ok.

I am interested in a mode to revert to the old system, even if data
quality will be worse.  My studies can run for months or a year, and
having the alarm notification not functional for that long is kind of
bad.  I'm adding a hardcoded bool resources that can turn it on and
off - but you'll have to check if it's useful or not.

I also thought to add a hard-coded resource that lets the scheduler
tick less often that every minute.  Not useful for everything, but if
someone wants to emphasize batter life over data exactness, then this
is useful.

I'll send a pull request you can look at.  Any ideas in meantime?

Thanks,

- Richard

rkdarst, 2017-07-14:
Does the alarm ticker only tick the scheduler (active parts), or does it also keep the rest of the passive data collection running?

denzilferreira, 2017-07-14:
Both.

Denzil;

closed 2017-07-18 by rkdarst