denzilferreira/aware-client #215: aware-phone: Frontend UI not always updated for server-provided config (for keys with dependency)

rkdarst, 2017-07-17:
Hi,

So we have a study doing a fairly normal thing:
1) AWARE configured on server
2) Phone is linked and server send JSON config to the client
3) Client stores config in `aware.db`, table `aware-settings`
4) When you use the client, you see the settings provided by the server.

I got a report that this wasn't working for some settings, and after investigation I see that it:
- *is* storing the server-side settings in the database
- The aware-phone UI doesn't update to reflect what is in aware.db
- This only happens for settings that have an `android:dependency` in `aware_preferences.xml`.
I hypothesize that this is related to Aware using the android-side preferences DB at the same time as it uses `aware.db` to store preferences, and android is trying to be smart about using the first one, but I don't know much more than that.

I have confirmed that things are put into the DB, and I *think* that this is enough to make Aware work anyway (**is this correct?**).

I also don't offhand know how to fix it, but I might look later.

denzilferreira, 2017-07-18:
The configuration on the client always matches the server configuration. Double check your server side keys => values and the aware_preferences.xml. The UI simply replicates what is stored on the aware_preferences.xml (if manually toggled via UI) and updates the local database (aware_settings). We backup the values to this database to allow access to the current configuration to other applications/plugins (Aware.setSetting() call).

`public void onSharedPreferenceChanged(SharedPreferences sharedPreferences, String key) {
        String value = "";
        Map<String, ?> keys = sharedPreferences.getAll();
        if (keys.containsKey(key)) {
            Object entry = keys.get(key);
            if (entry instanceof Boolean)
                value = String.valueOf(sharedPreferences.getBoolean(key, false));
            else if (entry instanceof String)
                value = String.valueOf(sharedPreferences.getString(key, "error"));
            else if (entry instanceof Integer)
                value = String.valueOf(sharedPreferences.getInt(key, 0));
        }

        Aware.setSetting(getApplicationContext(), key, value);
        ...
`

rkdarst, 2017-07-18:
I did check the keys, but I'll do it again once I see that phone.  I know they *should* be synced, but still I saw a client with server-side config that looked correct. the value set correctly in the raw `aware_settings` sqlite database when I dumped it, but not set correctly in the UI.  Very strange...

denzilferreira, 2017-10-31:
This should be in sync now. Can you confirm?

denzilferreira, 2019-10-01:
Seems to be OK.

closed 2019-10-01 by denzilferreira