denzilferreira/aware-client #231: Using the Aware Library but can't find symbol class Aware

kyleesantos, 2017-10-26:
I am trying to create a standalone application in android studio that uses Aware as a built-in library. I followed the step-by-step process on the Aware framework site and it seemed to work until the last step of actually using the sensors. I do not need a plugin for my application, I just want to access the sensor data directly. I tried putting the sample code for accessing accelerometer data in my MainActivity class but it is raising errors that it cannot find the symbol class Aware or the variables Aware_preferences. Do I have to import the Aware.java code in the github in order for it to successfully build? Or does using Aware as the built-in library allow me to skip that step?

denzilferreira, 2017-10-31:
Make sure your project has downloaded the library to your project. If it can't find the classes, it's because the library is not yet on your project.

kmarkiv, 2018-11-09:
Hi @denzilferreira I seem to have the same issue. I have added the project on gradle/manually but it says package com.aware not found. I'm able to import other libraries though. 

[image]

denzilferreira, 2018-11-09:
Make sure to be targeting SDK 28 on your app.

kmarkiv, 2018-11-09:
HI @denzilferreira, 

I created a sample project in API 28.  Added aware and a few dependencies mentioned here. 

http://www.awareframework.com/creating-a-standalone-application/

[image]

Let me know if there is an updated tutorial on this and I'll try to follow it. I am not a regular Android developer so I might be missing something. Thanks in advance.

denzilferreira, 2019-10-01:
See README for how to start with library on a standalone app.

closed 2019-10-01 by denzilferreira