denzilferreira/aware-client #240: Notifications not working on Android 8 or above

WorldofJARcraft, 2018-03-14:
Android 8 requires every notification to be mapped to a particular notification channel, else it will not be displayed. Because of this bevaviour, no notifications about pending questionnaires or similar things will display. The easiest solution would be to create a notification channel in the Aware class, save its ID in the Aware_Preferences class and use it in every class that displays notifications.
One could also create several different channels, e.g. one entire channel only for ESM questionnaires, another one for warnings and so on, every one of them with suitable settings (light, sound, ...).
Finally, make sure to run all notification channel related code only on Android 8 or above, because it will not work on Android 7 or lower.

denzilferreira, 2018-03-14:
Hi, this is solved in the development branch :)

WorldofJARcraft, 2018-03-14:
Thank you, I work with master branch and did not notice it. I will clode the issue.

closed 2018-03-14 by WorldofJARcraft