denzilferreira/aware-client #241: Problem with implicit broadcasts on Android 8 or higher

WorldofJARcraft, 2018-03-14:
I was using the MQTT service of an Android 8 device to remotely trigger ESM questionnaires. Although the MQTT services worked as they should, an notification about a pending questionnaire would not display (see Issue #240 ). I fixed the notification method, but I ran into another problem. The class ESMMonitor is registered as receiver for ESM-related broadcasts in the Application Manifest. 
Android 8 will only allow this kind of declaration for a few internal broadcast messages like BOOT_COMPLETED. ESMMonitor hast to be registered as receiver programmatically, e.g. in the Aware.startAware() method.

denzilferreira, 2018-03-14:
Are you using the development branch? If you are using MQTT to trigger the ESM, make sure the ESM sensor is enabled so you can see the notifications when receiving the MQTT message on the esm topic your device subscribes to.

WorldofJARcraft, 2018-03-14:
No, I am using master branch. Possibly this issue has been solved already on development.
ESM sensor is enabled and the broadcast from MQTT seems correct. This is certainly not the problem. I used a debugger to confirm that the ESMMonitor.onReceive() Method itself is not called. When I remember correctly, the check if ESM is enabled is done inside this method. When I register ESMMonitor as receiver programmatrically, everything works as it should.

commit b1dddc3e (2018-09-05): - Fixed #241: removing implicit broadcasts that are no longer available in Android 8. replaced with run-time broadcastreceiver registration.
closed 2018-09-05 by denzilferreira