denzilferreira/aware-client #242: Download of certificates not working on any testing device featuring Android or higher

WorldofJARcraft, 2018-03-14:
Currently, I am struggling to set up an own instance of AWARE Dashboard. When trying to connect a demo app to this server, it would only create ANR's and crashes. I tracked the problem down to the SSLManager class. 
First of all, any attempts to connect to a webserver have to be run in an Async task which is not the case here.
Second, your approach to retrieve the SSL certificate seems highly doubtful. Correct me if I'm wrong, but I believe you are creating an Instance of HttpsUrlConnection, connecting it to the custom server and finally retrieving the server certificate from the HttpsUrlConnection instance. On all of my test devices, HttpsUrlConnection refuses to connect to a server if it does not offer a trusted certificate, so you will never be able to download and add a self-signed certificate. However, if the study server has a trusted certificate (e.g from Let'sEncrypt), there is no need to download it because it will automatically be accepted and can be used for connections right away in the Http/Https classes without any further actions.
My advice would be to completely remove the SSLManager class that currently only leads to crashes.

rkdarst, 2018-03-14:
For studies I do, we need to use SSL cert pinning.  So for that, this is useful and I've done some work on SSLmanager in the past.  Downloading the cert over non-ssl is strange, but for our own projects, we adopted a system of including the sha256 of the cert in the QRcode, so that the cert is also validated and no other trust is required.

What version of android do you have on the test devices?

In my study, I include the crt_url parameter in the QR code so that I can direct the cert be downloaded over a HTTPS connection with a valid cert, then pinned via sha256 as above.  However, I have set up my own server to do all of this, and I don't think anyone else uses this yet.  The required code are in upstream Aware, though.

WorldofJARcraft, 2018-03-14:
I observed the following behaviour on an Android 4.4, an Android 5.1, an Android 7.1 and an Android 8.1 device: When I install the Aware Phone app, it keeps crashing or sending me App not responding signals all the time. So I created a demo app and included aware-core as a module. I noticed that the ANR's are related to the SSLManager class. So I set up a simple Apache webserver with self-signed certificate and tried to download and install its certificate. This did not work, because HttpsUrlConnection refused the connection before I was able to retrieve the certificate from it.

WorldofJARcraft, 2018-05-23:
We set up an AWARE test server on a local server. Whenever we tried to join a study, there was an error message telling us that study information cannot be retrieved.
As I investigated the problem, I found it's origin in com.aware.utils.SSLManager, in the downloadCertificate Method to be exact. There, you call retrieveRemoteCertificate and hand over the hostname, but not the protocol. Because you cast url.openConnection() to a HttpsURLConnection, I assume that you presume the protocol is https. But you can not instanciate a URL without a protocol. 
In retrieveRemoteCertificate, there is a while loop waiting for the response code of the casted connection to change. As I see it, it never changes. You can simply state conn.connect(); and then try to retrieve the certificates.
Finally, here is my modified code for this two methods that perfectly works on my Android 8.0 test device:
```java
/**
     * Downloads the certificate directly from the URL, instead of a public folder.
     * @param url
     * @return
     */
    public static X509Certificate retrieveRemoteCertificate(URL url) {
        try {
            HttpsURLConnection conn = (HttpsURLConnection) url.openConnection();
            conn.setConnectTimeout(5000); //5 seconds to connect
            conn.setReadTimeout(10000); //10 seconds to acknowledge the response

            long now = System.currentTimeMillis();
            /*while (conn.getResponseCode() != HttpsURLConnection.HTTP_OK || now - System.currentTimeMillis() <= 5000) {
                //noop - wait up to 5 seconds to retrieve the certificate
            }*/

            conn.connect();
            // retrieve the N-length signing chain for the server certificates
            // certs[0] is the server's certificate
            // certs[1] - certs[N-1] are the intermediate authorities that signed the cert
            // certs[N] is the root certificate authority of the chain
            Certificate[] certs = conn.getServerCertificates();
            if (certs.length > 0 && certs[0] instanceof X509Certificate) {
                // certs[0] is an X.509 certificate, return its "notAfter" date
                return ((X509Certificate) certs[0]);
            }

            // connection is not HTTPS or server is not signed with an X.509 certificate, return null
            return null;
        } catch (SSLPeerUnverifiedException spue) {
            // connection to server is not verified, unable to get certificates
            Log.d(Aware.TAG, "Certificates: " + spue.getMessage());
            return null;
        } catch (IllegalStateException ise) {
            // shouldn't get here -- indicates attempt to get certificates before
            // connection is established
            Log.d(Aware.TAG, "Certificates: " + ise.getMessage());
            return null;
        } catch (IOException ioe) {
            // error connecting to URL -- this must be caught last since
            // other exceptions are subclasses of IOException
            Log.d(Aware.TAG, "Certificates: " + ioe.getMessage());
            return null;
        }
    }
/**
     * Classic method: Download certificate unconditionally.  This is the old
     * method of certificate fetching.  This method blocks if the block parameter is true,
     * otherwise is nonblocking.
     *
     * @param context  app contexto
     * @param hostname Hostname to download.
     * @param block    If true, block until certificate retrieved, otherwise do not.
     */
    public static void downloadCertificate(Context context, String protocol, String hostname, boolean block) {
        //Fixed: make sure we have a valid hostname
        if (hostname == null || hostname.length() == 0) return;

        // api.awareframework.com is an exception: we download from a different host.
        // cert_host is the host from which we actually download.
        String cert_host;
        if (hostname.contains("api.awareframework.com")) {
            cert_host = "awareframework.com";
        } else cert_host = hostname;

[... 29 lines cut ...]
                            if (e != null) {
                                Log.d(Aware.TAG, "ERROR SSL certificate: " + e.getMessage());
                            }
                        }
                    });

            if (block) {
                try {
                    https.get();
                } catch (java.lang.InterruptedException | ExecutionException j) {
                    // What to do here?
                }
            }
        }
    }
```
I determine the protocol using
```java
String protocol = "http";
        try {
            protocol = new URL(url).getProtocol();
        } catch (MalformedURLException e) {
            e.printStackTrace();
        }
```

Helma-T, 2018-08-08:
Hi, Thank you for your input. It works

commit adf305f8 (2018-09-05): - Fixed #242: thanks to @WorldofJARcraft
- SSL certificates update automatically when not matching the study (expired, transitioned server).
closed 2018-09-05 by denzilferreira