denzilferreira/aware-client #251: Unable to resolve dependency when opening Project

s3and0s, 2018-12-19:
I ran into this Gradle Build sync error, with unresolved dependencies. When clicking "Open file" under the error message, it brings up build.gradle (module: aware-phone). 

I'm not sure how to resolve and run the build successfully. Please let me know what other information to provide to assist the resolution of this issue. 

[image]

Here is the full error message from the failed sync build: 
Unable to resolve dependency for ':aware-phone@debug/compileClasspath': Could not resolve project :com.aware.plugin.device_usage.
Open File
Show Details

Unable to resolve dependency for ':aware-phone@debug/compileClasspath': Could not resolve project :com.aware.plugin.ambient_noise.
Open File
Show Details

Unable to resolve dependency for ':aware-phone@debugAndroidTest/compileClasspath': Could not resolve project :com.aware.plugin.device_usage.
Open File
Show Details

Unable to resolve dependency for ':aware-phone@debugAndroidTest/compileClasspath': Could not resolve project :com.aware.plugin.ambient_noise.
Open File
Show Details

Unable to resolve dependency for ':aware-phone@debugUnitTest/compileClasspath': Could not resolve project :com.aware.plugin.device_usage.
Open File
Show Details

Unable to resolve dependency for ':aware-phone@debugUnitTest/compileClasspath': Could not resolve project :com.aware.plugin.ambient_noise.
Open File
Show Details

Unable to resolve dependency for ':aware-phone@release/compileClasspath': Could not resolve project :com.aware.plugin.device_usage.
Open File
Show Details

Unable to resolve dependency for ':aware-phone@release/compileClasspath': Could not resolve project :com.aware.plugin.ambient_noise.
Open File
Show Details

Unable to resolve dependency for ':aware-phone@releaseUnitTest/compileClasspath': Could not resolve project :com.aware.plugin.device_usage.
Open File
Show Details

Unable to resolve dependency for ':aware-phone@releaseUnitTest/compileClasspath': Could not resolve project :com.aware.plugin.ambient_noise.
Open File
Show Details

denzilferreira, 2018-12-19:
Mea culpa. I'll fix it!

denzilferreira, 2018-12-19:
Should be fixed in latest 👍

closed 2018-12-19 by denzilferreira