denzilferreira/aware-client #268: Google Activity Plugin not sampling according to set frequency

nleo575, 2019-10-10:
Issue found on build 256.

This plugin does't appear to be sampling at the set frequency. During a 3 week test, the sensor was set to sample every 60 seconds. However, Android devices sampled anywhere between 1 activity per hour, and 372 activities per hour. Oversampling continued despite us turning on "enforce frequency of all sensors" under the advanced options. 

Issue rating: Minor - 2 out of 5 where 5 is worse.

denzilferreira, 2019-10-18:
This is controlled by google services. We just tell google the rate in which we'd like to know the activity but then other factors from Google, like if you have moved or not, used the phone or not, etc. will trigger this. This is beyond our control.

nleo575, 2019-10-21:
@denzilferreira what does the "enforce frequencies of all sensors" actually do/work on then?

denzilferreira, 2019-10-22:
This is only regarding the high-sampling sensors: accelerometer, gyroscope, barometer, gravity, linear accelerometer, magnetometer, rotation, temperature (https://github.com/denzilferreira/aware-client/blob/master/aware-core/src/main/java/com/aware/syncadapters/AwareSyncAdapter.java#L62). For these sensors, the sensormanager reports according to the milliseconds delay. But if you want to enforce a specific sampling rate across all of them, it will downsample or upsample depending on the limitations of the sensors on your device.

closed 2019-10-30 by denzilferreira