denzilferreira/aware-client #269: App overrides dashboard settings for some sensors

nleo575, 2019-10-10:
Issue found on build 256.

The following sensors overrode dashboard settings after joining the study using a QR code. We were able to manually go in a configure each setting in the app to the correct value though. 

* Accelerometer: dashboard setting = 1000ms, app overrode to 200ms
* Bluetooth: dashboard setting = 300s, app overrode to 60s
* Google Fused location: dashboard was set to High/102 (GPS+network+WiFi), app overrode to medium (network + WiFi)
* Light: dashboard = 1 second (1E6 microseconds), app overrides to default 2E5 microseconds
* Locations: dashboard setting = 300s, app overrode to 180s
* Mobile ESM/EMA: turned off completely on dashboard, app overrode and turned it on
* MQTT server: dashboard turned off, app overrode and turned it on

Issue rating: Major - 4 out of 5 where 5 is worse.

denzilferreira, 2019-10-18:
The server side settings (when set) always take priority over the app settings (this avoids participants fiddling with the settings). If you want to allow changes on the app, then set no settings on the dashboard. Need to check if this is still the case.

nleo575, 2019-10-21:
@denzilferreira I realize that's the expected behavior (sever takes precedence over app), which is exactly what we saw with the iOS devices. On all of the Androids the server settings were being ignored and the app was taking the default settings.

I tried to test this today using the latest build, but 259 won't let me join any studies using URL or QR code.

JulioV, 2020-04-01:
Fixed in 54fe2e64d335da7120a18edde6d0533fa3582b60, except for plugins (although the correct settings are being stored)

closed 2020-04-01 by JulioV