denzilferreira/aware-client #270: Timezone sensor not sampling according to set frequency

nleo575, 2019-10-10:
Issue found on build 256.

This plugin does't appear to be sampling at the set frequency. During a 3 week test, the sensor was set to sample every 7200 seconds. However, Android devices sampled anywhere between 1 time per hour, and 20 times per hour. Oversampling continued despite us turning on "enforce frequency of all sensors" under the advanced options.

Issue rating: Minor - 2 out of 5 where 5 is worse.

denzilferreira, 2019-10-18:
The timezone is event based. It should only log if you change timezones.

nleo575, 2019-10-21:
Then why is there a setting to set the sampling frequency if it is event based? Also none of our participants left our timezone but as I mentioned it was drastically oversampling.