denzilferreira/aware-client #272: Unable to join study using URL

nleo575, 2019-10-10:
Issue found on build 256.

We were unable to join a study by pasting a URL into the app itself. After clicking activate to turn on data collection, the app brings you back to the home screen with no change. The only way to join the study appears to be using the QR code. 

Issue rating: Minor - 2 out of 5 where 5 is worse.

denzilferreira, 2019-10-18:
Hey, can you try this sequence:
- put URL
- enable Active on study
- set the settings you want to enable data collection

That should work with the study URL that is provided by the dashboard interface (not the browser URL because it's missing the API key)

nleo575, 2019-10-21:
@denzilferreira 

Update: Using the latest build (259) I can't join any study using the URL or the QR code. 

I'm getting this error in the app on two different studies and using two different Android phones:
[image]

nleo575, 2019-10-21:
@denzilferreira I just downloaded and tried with build 258 and it doesn't have either URL or QR code issue. So to recap: URL joining issue in 256, No issues in 258, Both URL & QR issue in 259.

commit 05450921 (2019-10-23): - Close #272: Fixed joining by QRCode and Aware.joinStudy(studyURL)
commit e54bb098 (2019-10-29): - Close #272: Fixed joining by QRCode and Aware.joinStudy(studyURL)
- Simplify dependencies across library and source code
-
closed 2019-10-23 by denzilferreira