denzilferreira/aware-client #273: Sensors' details screen doesn't reflect current study settings

nleo575, 2019-10-10:
Issue found on build 256.

Current app state (i.e. specific settings) aren't displayed when the user clicks on a sensor's name from the main screen. Instead, on several sensor detail screens there are either dummy values (e.g. X seconds) or the app only displays the default value (e.g. 0 seconds) instead of the real setting. 

Noticed this on the following sensors/plugins:
* Bluetooth: X seconds
* Fused location: always displays location accuracy as "medium" but if you click on the setting, it displays the correctly set value within the setting selection popup. 
* Location: displays generic units instead of current settings

Issue rating: Minor - 2 out of 5 where 5 is worse.

denzilferreira, 2019-10-18:
Thanks, will check if we are setting the default values.

nleo575, 2019-10-21:
Also noticed this on Processor workload setting's page. Displays "Every X seconds".

JulioV, 2020-04-01:
Fixed in 54fe2e64d335da7120a18edde6d0533fa3582b60

closed 2020-04-01 by JulioV