denzilferreira/aware-client #292: Custom Aware client manual syncing not working

xubowenhaoren, 2020-05-22:
Hello, 

We are trying to build a Cordova plugin based on the latest release of Aware Android, `com.github.denzilferreira:aware-client:master-SNAPSHOT`.

I confirm that the study server is running: 
```
"intent_action":"ACTION_AWARE_KEEP_ALIVE","class":"edu.berkeley.eecs.emission\/com.aware.Aware"},"trigger":{"interval":30,"random":{},"context":[],"condition":[]}}}
2020-05-22 11:07:30.798 18362-18691/edu.berkeley.eecs.emission D/AwarePlugin- Aware: Study info: {
         "study_name": "EmissionV2",
         "study_description": "The 2nd study demo for emission",
         "researcher_first": "Bowen",
         "researcher_last": "Xu",
         "researcher_contact": "****.edu"
    }
2020-05-22 11:07:31.597 18362-18691/edu.berkeley.eecs.emission D/AwarePlugin- Aware: Study config: [
         {
              "sensors": [
                   {
                        "setting": "status_battery",
                        "value": "true"
                   },
                   {
                        "setting": "status_bluetooth",
                        "value": "true"
                   },
                   {
                        "setting": "status_location_gps",
                        "value": "true"
                   },
                   {
                        "setting": "status_location_network",
                        "value": "true"
                   },
                   {
                        "setting": "status_network_events",
                        "value": "true"
                   },
                   {
                        "setting": "status_wifi",
                        "value": "true"
                   },
                   {
                        "setting": "status_mqtt",
                        "value": "true"
                   },
                   {
                        "setting": "mqtt_server",
                        "value": "YOUR_MOSQUITTO_IP"
                   },
                   {
                        "setting": "mqtt_port",
                        "value": "1883"
                   },
                   {
                        "setting": "mqtt_keep_alive",
                        "value": "600"
                   },
                   {
                        "setting": "mqtt_qos",
                        "value": "2"
                   },
                   {
                        "setting": "status_esm",
                        "value": "true"
                   },
                   {
                        "setting": "mqtt_username",
                        "value": "7c940f31-XXXX-XXXX-XXXX-6f8a9ad2566a"
                   },
                   {
                        "setting": "mqtt_password",
                        "value": "7Fb8EOPdoauS"
                   },
                   {
                        "setting": "study_id",
                        "value": "1704"
                   },
                   {
                        "setting": "study_start",
                        "value": "1590170852049"
                   },
                   {
                        "setting": "webservice_server",
                        "value": "https:\/\/****.cs.washington.edu\/index.php\/webservice\/index\/****\/****J8RjTvlf"
                   },
                   {
                        "setting": "status_webservice",
                        "value": "true"
                   }
              ]
         }
    ]
```

I had turned on Aware debugging: 
```java
        Aware.setSetting(cordova.getActivity().getApplicationContext(),
                Aware_Preferences.DEBUG_FLAG, "true");
        Aware.setSetting(cordova.getActivity().getApplicationContext(),
                Aware_Preferences.DEBUG_TAG, TAG + "- Aware");
```

However, when I tried to manual sync, 
```java
    class SyncNowTask extends AsyncTask<Void, Void, Void>
    {
        @Override
        protected Void doInBackground(Void... params) {
            Intent syncNow = new Intent(Aware.ACTION_AWARE_SYNC_DATA);
            ctxt.sendBroadcast(syncNow);
            return null;
        }
    }
```

The app (or the plugin) didn't respond to sync or produce an error log. I noticed that in [Aware.java](https://github.com/denzilferreira/aware-client/blob/master/aware-core/src/main/java/com/aware/Aware.java#L2189), The syncing mechanism uses `ContentResolver.requestSync`. After reading a StackOverflow page, I realized that I might be missing some XML config in my plugin. 

Could you point out what am I missing? Thanks!

denzilferreira, 2020-05-23:
Hi, you are missing to add your plugin contentprovider as a syncadapter (see xml/) so that Android knows it is there and available to sync with the other sensors and plugins. See: https://github.com/denzilferreira/aware-client/blob/master/aware-core/src/main/res/xml/sync_accelerometer.xml

Notice that for this to work, you need to create and sign your own AWARE client, as syncadapters only work for the same accountType and developer certificate. In the future, we're abandoning syncadapters altogether because of this (this essentially broke the plugin-based architecture and we have to bundle all plugins together with the client).

xubowenhaoren, 2020-05-26:
> you need to create and sign your own AWARE client

Do you mean building the Cordova app with my own developer key? 

> In the future, we're abandoning syncadapters altogether because of this

Can I still rely (for the lack of a better choice) on the syncadapters now?

denzilferreira, 2020-05-26:
Yes to both questions :)

#> sudo mail from:Denzil
#> Sent successful (elapsed .001 seconds)

xubowenhaoren, 2020-05-28:
Hello, 

I added all XML's in your `aware-core/src/main/res/xml` folder to my plugin and set up an appropriate target directory redirect as well: 

```
        <resource-file src='res/android/xml/sync_gyroscope.xml' target='res/xml/sync_gyroscope.xml'/>
        <resource-file src='res/android/xml/sync_applications.xml' target='res/xml/sync_applications.xml'/>
        <resource-file src='res/android/xml/aware_preferences.xml' target='res/xml/aware_preferences.xml'/>
        ...
        <resource-file src='res/android/xml/sync_timezones.xml' target='res/xml/sync_timezones.xml'/>
```

But checking the logcat output, the only output is this: 

```
2020-05-28 11:49:55.824 26155-26613/edu.berkeley.eecs.emission D/AwarePlugin: manual syncing...
2020-05-28 11:49:55.834 26155-26613/edu.berkeley.eecs.emission D/AwarePlugin: Account {name=awareframework, type=com.awareframework}
2020-05-28 11:49:55.840 26155-26613/edu.berkeley.eecs.emission D/AwarePlugin: edu.berkeley.eecs.emission.provider.aware
```

So I still don't know whether the manual sync succeeded or not. I remember that in a standard Aware Android client is triggered, a long list of data uploading should be in the logcat. Is there anything else that I am missing?

xubowenhaoren, 2020-06-23:
Hello Denzil, 

In my fork, I added logcats that reveal the Aware-as-plugin background tasks; I compared them with the native Aware Android client but didn't find anything revealing. Could you take a look and point out what's missing? The link is [here](https://github.com/e-mission/e-mission-docs/issues/501#issuecomment-645851197).

xubowenhaoren, 2020-07-12:
See https://github.com/e-mission/e-mission-docs/issues/501. Issue solved.

closed 2020-07-12 by xubowenhaoren