denzilferreira/aware-client #299: Android 12+ support

santaCruz1854789, 2022-05-05:
Good evening,
when I set the compileSdkVersion to 31 in my gradle.build file the building process starts to act up.
Since "Apps targeting Android 12 and higher are required to specify an explicit value for `android:exported`" it tells me "Manifest merger failed : android:exported needs to be explicitly specified for element <service#com.aware.utils.Aware_Accounts>" and also "android:exported needs to be explicitly specified for element <service#com.aware.utils.Aware_Accounts>.", those are lines I was not able to find in the AndroidManifest.xml file and thus led me to believe that they were in the source code (i.e. out of my reach), is that correct?
If so, will you change the code to specify exported/give us a way to do so?

Thank you very much.

santaCruz1854789, 2023-04-07:
@denzilferreira Please could you look into this?

santaCruz1854789, 2023-04-07:
@denzilferreira Could you please look into this request? 
 https://github.com/denzilferreira/aware-client/blob/master/aware-core/src/main/AndroidManifest.xml
 Line 44.
 With the latest build it's now a necessity to specify exported for this service ( and I suppose the other ones as well).
The docs: https://developer.android.com/guide/topics/manifest/activity-element#exported

denzilferreira, 2023-04-07:
What the error is saying is that those values are not explicitly set. You can add them to AndroidManifest.xml on your app (copy the item you want to override to your app manifest and when it merges, it uses your version).

santaCruz1854789, 2023-04-07:
Yeah, I ended up doing just that.
I was meaning to fork your code and do a pull request, just letting you know.